jayofelony/pwnagotchi #278: Help with Macchanger on Pwnagotchi

Dinarova, 2024-12-26:
Hello everyone,

This is not an issue, but a question. How can I change the MAC address of the external USB Wi-Fi interface? Previously, I modified the file using sudo nano /etc/rc.local, adding the line:

/sbin/ifconfig wlan0mon down; /usr/bin/macchanger -r wlan0mon; /sbin/ifconfig wlan0mon up

However, this file no longer exists in the current Pwnagotchi image. If anyone can help me, I would appreciate it.
 
I know I can do it manually, but I want to do it automatically on every boot. 
Thanks!

wpa-2, 2024-12-26:
Last time I installed MAC changer it gave me the option to change on boot. 

[image]

Dinarova, 2024-12-26:
> Last time I installed MAC changer it gave me the option to change on boot.
> 
> ![2024-12-26 18_45_18-Raspberry Pi MAC Address Spoofing - Pi My Life Up - Opera](https://private-user-images.githubusercontent.com/9049886/398786341-c4af7e6b-b40b-477c-b2fa-5f8daa305ab4.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzUyNDEwMTIsIm5iZiI6MTczNTI0MDcxMiwicGF0aCI6Ii85MDQ5ODg2LzM5ODc4NjM0MS1jNGFmN2U2Yi1iNDBiLTQ3N2MtYjJmYS01ZjhkYWEzMDVhYjQucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTIyNiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEyMjZUMTkxODMyWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YTAyZjZmNGI3ZGVkMTM1ZmIzMTg1M2MzMzczYjBiZjU2ZTBmMDI1MmJiZmE2ODZjYzFkNzdkOWFmYjFiNDhiMyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.qOUj1is9f0OE7UV-V0L8a7nLHwU8FkIHfs2XFt1SDOA)

Yeah I had that screen went installing macchanger but it doesn't change it automatically.  I tried many scripts and plugin but just change wlan0 and not on wlan0mon

jayofelony, 2024-12-26:
I've answered on reddit..

jayofelony, 2024-12-28:
Closing as this is not really related to Pwnagotchi software.

closed 2024-12-28 by jayofelony