jayofelony/pwnagotchi #281: [Bug]: Newly flashed 2.9.3-2 pwnagotchi --wizard is nonfunctional

RollPitchYall, 2024-12-28:
### Contact Details

matthewarch314@gmail.com

### What happened?

Problem: 
pwnagotchi returns: -bash: pwnagotchi: command not found
sudo pwnagotchi returns the error: sudo: pwnagotchi: command not found,
pwnlog returns: sudo: pwnlog: command not foun
How to reproduce: 
flash the [pwnagotchi-64bit.img.xz](https://github.com/jayofelony/pwnagotchi/releases/download/v2.9.3-2/pwnagotchi-64bit.img.xz) file onto a (arguably cheap) 64 gb microsd card using raspberry pi imager, and run it on a raspberry pi 0 2w

[image]

### Version

2.9.3

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wpa-2, 2024-12-28:
You are going to need to reflash, the wiki clearly states using pi user only.

RollPitchYall, 2024-12-28:
thank you, I was following along using a different tutorial.

wpa-2, 2024-12-28:
> thank you, I was following along using a different tutorial.

No problem, if you sort it can you close this please. 
Good luck..

RollPitchYall, 2024-12-28:
I reflashed following the tutorial, and your advice, leaving customization settings off and things are now progressing as expected. Closing the issue now

closed 2024-12-28 by RollPitchYall