jayofelony/pwnagotchi #284: [Bug]:  invalid setting: Key file contains key “autoconnect”

Delurkdotcom, 2024-12-31:
### What happened?

While configuring my default internal interface from **wlan0** to **wlan1**, I noticed a configuration issue in **NetworkManager** with the following error:

```
keyfile: connection.autoconnect: invalid setting: Key file contains key “autoconnect” which has a value that cannot be interpreted.
```

It turns out that the **usb0.nmconnection** file is misconfigured. The existing key `autoconnect=yes` should be changed to `autoconnect=true` to avoid this warning. Otherwise, why was this choice made in the first place?

### Version

2.9.3-2

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

jayofelony, 2024-12-31:
[image]

Yeahhhh, no.

Delurkdotcom, 2024-12-31:
Yes, I have already looked at the possible keyfiles, but the warning is still there. However, you set true for the eth0 interface and not for usb0. Why this difference? Moreover, depending on the version of the program (in this case NetworkManager), certain arguments may no longer be recognized.

Delurkdotcom, 2024-12-31:
Could you check your logs, for example by running `journalctl -u NetworkManager`, to see if the warning appears the

jayofelony, 2024-12-31:
I will later, working now. What happens if you set them all to either true or yes.

It was a PR done by someone else.

jayofelony, 2025-01-01:
I commited the usb0 file with true instead of yes. I have tried every possible combo but 'yes' is apparently not accepted for some reason even though docs say it can.

closed 2025-01-01 by jayofelony