jayofelony/pwnagotchi #285: Suggestion for Release Naming Convention

drego85, 2025-01-02:
Hi,
I would like to propose an improvement to the naming convention of the release version for Pwnagotchi.

Currently, any Pwnagotchi version is released with names like `pwnagotchi-64bit.img.xz`. I believe that including the released version in the file name could make it easier for users to quickly identify which version they are downloading.

For example, I suggest changing the file name format to something like `pwnagotchi-64bit_2.9.4.img.xz`, where `2.9.4` represents the specific release version. This change could enhance usability and reduce confusion, especially for users who may need to download specific versions for compatibility or testing purposes.

Thank you for considering this proposal!

jayofelony, 2025-01-02:
Yes it got lost at some point, I've been forgetting that.

SgtStarSlayer, 2025-01-02:
not to throw a wrench in your update process, not sure if the ups 1.3 plugin is working properly. not getting a reading at all. it was working back in 2.8.3 for a short period then just stopped. back then i just jumped on that iteration.

MimoAW, 2025-01-22:
> not to throw a wrench in your update process, not sure if the ups 1.3 plugin is working properly. not getting a reading at all. it was working back in 2.8.3 for a short period then just stopped. back then i just jumped on that iteration.

What version are you on?