jayofelony/pwnagotchi #286: [Bug]: Error in the pwnlog with Bettercap

Delurkdotcom, 2025-01-03:
### What happened?

I am using an Alfa AWUS036AXM USB interface (a/b/g/n/ac/ax), and I’ve noticed that Bettercap doesn’t seem to support all channels. It displays an error for channels that are not recognized. Is this the expected behavior of Bettercap?

Additionally, since I started using the Alfa adapter, my pwn counter remains stuck at zero.

`20:58:17 [INFO] [Thread-7] : Active: [1, 3, 6, 11, 100, 149], Next scan: [1, 3, 6, 11, 100, 149, 29, 4, 181], yet unscanned: 95 [1, 2, 3, 5, 6, 7, 8, 9, 10, 11, 36, 40, 44, 48, 52, 56, 60, 64, 100, 104, 108, 112, 116, 120, 124, 128, 132, 136, 140, 144, 149, 153, 157, 161, 165, 169, 173, 177, 1, 5, 9, 13, 17, 21, 25, 33, 37, 41, 45, 49, 53, 57, 61, 65, 69, 73, 77, 81, 85, 89, 93, 97, 101, 105, 109, 113, 117, 121, 125, 129, 133, 137, 141, 145, 149, 153, 157, 161, 165, 169, 173, 177, 185, 189, 193, 197, 201, 205, 209, 213, 217, 221, 225, 229, 233]
20:58:18 [INFO] [MainThread] : error 400: 181 is not a valid wifi channel
20:58:18 [ERROR] [MainThread] : Error while setting wifi.recon.channels (error 400: 181 is not a valid wifi channel)
(most recent call last):
 File "/usr/lib/python3/dist-packages/requests/models.py", line 971, in json
   return complexjson.loads(self.text, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
 File "/usr/lib/python3.11/json/__init__.py", line 346, in loads
   return _default_decoder.decode(s)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^
 File "/usr/lib/python3.11/json/decoder.py", line 340, in decode
   raise JSONDecodeError("Extra data", s, end)
Extra data: line 1 column 5 (char 4)

handling of the above exception, another exception occurred:

(most recent call last):
 File "/home/pi/.pwn/lib/python3.11/site-packages/pwnagotchi/bettercap.py", line 24, in decode
   return r.json()
          ^^^^^^^^
 File "/usr/lib/python3/dist-packages/requests/models.py", line 975, in json
   raise RequestsJSONDecodeError(e.msg, e.doc, e.pos)
Extra data: line 1 column 5 (char 4)

handling of the above exception, another exception occurred:

(most recent call last):
 File "/home/pi/.pwn/lib/python3.11/site-packages/pwnagotchi/agent.py", line 160, in recon
   self.run('wifi.recon.channel %s' % ','.join(map(str, channels)))
 File "/home/pi/.pwn/lib/python3.11/site-packages/pwnagotchi/bettercap.py", line 118, in run
   return decode(r, verbose_errors=verbose_errors)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
 File "/home/pi/.pwn/lib/python3.11/site-packages/pwnagotchi/bettercap.py", line 32, in decode
   raise Exception(err)
error 400: 181 is not a valid wifi channel
20:59:19 [INFO] [Thread-7] : Histo: {'loops': 31, 1: 31, 3: 31, 6: 201, 11: 97, 100: 62, 149: 31}
20:59:25 [INFO] [MainThread] : [epoch 31] duration=00:01:07 slept_for=00:01:00 blind=0 sad=0 bored=0 inactive=9 active=0 peers=0 tot_bond=0.00 avg_bond=0.00 hops=6 missed=0 deauths=0 assocs=0 handshakes=0 cpu=3% mem=40% temperature=44C reward=-0.053674892703862664
20:59:25 [INFO] [Thread-7] : Active: [1, 3, 6, 11, 100, 149], Next scan: [1, 3, 6, 11, 100, 149, 209, 61, 140], yet unscanned: 86 [2, 5, 7, 8, 9, 10, 36, 40, 44, 48, 52, 56, 60, 64, 104, 108, 112, 116, 120, 124, 128, 132, 136, 144, 153, 157, 161, 165, 169, 173, 177, 1, 5, 9, 13, 17, 21, 25, 33, 37, 41, 45, 49, 53, 57, 65, 69, 73, 77, 81, 85, 89, 93, 97, 101, 105, 109, 113, 117, 121, 125, 129, 133, 137, 141, 145, 149, 153, 157, 161, 165, 169, 173, 177, 185, 189, 193, 197, 201, 205, 213, 217, 221, 225, 229, 233]
20:59:25 [INFO] [MainThread] : error 400: 209 is not a valid wifi channel
20:59:25 [ERROR] [MainThread] : Error while setting wifi.recon.channels (error 400: 209 is not a valid wifi channel)
(most recent call last):
 File "/usr/lib/python3/dist-packages/requests/models.py", line 971, in json
   return complexjson.loads(self.text, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
 File "/usr/lib/python3.11/json/__init__.py", line 346, in loads
   return _default_decoder.decode(s)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^
 File "/usr/lib/python3.11/json/decoder.py", line 340, in decode
   raise JSONDecodeError("Extra data", s, end)
Extra data: line 1 column 5 (char 4)

handling of the above exception, another exception occurred:

(most recent call last):
 File "/home/pi/.pwn/lib/python3.11/site-packages/pwnagotchi/bettercap.py", line 24, in decode
   return r.json()
          ^^^^^^^^
 File "/usr/lib/python3/dist-packages/requests/models.py", line 975, in json
   raise RequestsJSONDecodeError(e.msg, e.doc, e.pos)
Extra data: line 1 column 5 (char 4)

handling of the above exception, another exception occurred:

(most recent call last):
 File "/home/pi/.pwn/lib/python3.11/site-packages/pwnagotchi/agent.py", line 160, in recon
   self.run('wifi.recon.channel %s' % ','.join(map(str, channels)))
 File "/home/pi/.pwn/lib/python3.11/site-packages/pwnagotchi/bettercap.py", line 118, in run
   return decode(r, verbose_errors=verbose_errors)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
 File "/home/pi/.pwn/lib/python3.11/site-packages/pwnagotchi/bettercap.py", line 32, in decode
   raise Exception(err)
error 400: 209 is not a valid wifi channel`

### Version

2.9.4

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

jayofelony, 2025-01-04:
Can you show me output of:
`sudo /sbin/iw phy0 channels | grep ' MHz' | grep -v disabled | sed 's/^.*\[//g' | sed s/\].*\$//g`

jayofelony, 2025-01-04:
OK, I am making a PR to bettercap to add checking for 6GHz.

jayofelony, 2025-01-04:
https://github.com/bettercap/bettercap/pull/1162

closed 2025-01-04 by jayofelony