jayofelony/pwnagotchi #287: [Bug]: 6Ghz SSID's don't seem to be found

rsun0525, 2025-01-03:
### Contact Details

richard.d.sun@gmail.com

### What happened?

I purchased a Panda PAU0F AXE3000 USB adaptor to connect to Raspberry Pi, but can't see 6Ghz in pwnagotchi

### Version

2.9.4

### Relevant log output

```shell
In regular Raspberry Pi, I can run "sudo iwlist wlan0 scan" see my 6Ghz SSID's, but after changing it pwnagotchi, I can't see it, is your driver out of date?
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

nasiralamreeki, 2025-01-03:
Maybe related I've noticed on the NOAI builds Pwnagotchi started properly scanning for 5ghz/6ghz but isn't detecting either SSIDS and this caused a huge delay and slows scanning. While if I turn off my external antenna scanning speeds up and is more efficient.

rsun0525, 2025-01-03:
I can see 5Ghz SSID's just fine, not 6Ghz.

jayofelony, 2025-01-03:
No logs, no support. Logs are there for a reason, and they do show what channels are available.

rsun0525, 2025-01-03:
Sorry, forgot to attach the log.
[pwnagotchi.log](https://github.com/user-attachments/files/18304534/pwnagotchi.log)

rsun0525, 2025-01-03:
Is there a comparable cinnabd iwlist I can run on the pwnagotchi to get a list of SSID's it can see?  I run the command on regular Raspberry Pi OS and things come back, but when I run it on pwnagotchi, I get:

pi@pwnagotchi400:~ $ sudo iwlist wlan0 scan
wlan0     Interface doesn't support scanning : Network is down

jayofelony, 2025-01-04:
Use wlan0mon for that command. What does pwnlog say at the start what channels are available.

I don't know if i can fix it, I don't really can give support for 3rd party hardware.

MimoAW, 2025-01-04:
Does it work for 6 GHz on a computer?

jayofelony, 2025-01-04:
`sudo /sbin/iw phy0 channels | grep ' MHz' | grep -v disabled | sed 's/^.*\[//g' | sed s/\].*\$//g`

rsun0525, 2025-01-04:
pi@pwnagotchi400:~ $ sudo /sbin/iw phy0 channels | grep ' MHz' | grep -v disabled | sed 's/^.*\[//g' | sed s/\].*\$//g
1
2
3
4
5
6
7
8
9
10
11
12
13
14
36
40
44
48
52
56
60
64
100
104
108
112
116
120
124
128
132
136
140
144
149
153
157
161
165

jayofelony, 2025-01-04:
So apparently it only goes up to 5ghz, not 6?

MimoAW, 2025-01-04:
That should mean it's functioning in 6 GHz then

rsun0525, 2025-01-04:
pi@pwnagotchi400:~ $ sudo iw phy0 info > output.txt
[output.txt](https://github.com/user-attachments/files/18307143/output.txt)

I also have Acrylic WI-Fi Analyzer and I can see my 3 6Ghz's SSID's are set to Channel 47, 111, and 79.
[image]

jayofelony, 2025-01-04:
> pi@pwnagotchi400:~ $ sudo iw phy0 info > output.txt [output.txt](https://github.com/user-attachments/files/18307143/output.txt)
>
> I also have Acrylic WI-Fi Analyzer and I can see my 3 6Ghz's SSID's [...]

If you scroll down that list you can see that the 6ghz channels are disabled on your dongle.

rsun0525, 2025-01-04:
When I look at the output of iw, it doesn't list 6Ghz, just thinking the driver in pwnagotchi doesn't support 6Ghz, try this same test with the adaptor and regular Raspberry Pi and see what it shows.

MimoAW, 2025-01-04:
It definitely does

MimoAW, 2025-01-04:
[image]

jayofelony, 2025-01-04:
I also just noticed that Bettercap doesnt seem to work with 6GHz yet, I just made a PR for that. So until then we have to do with 2.4/5GHz.