jayofelony/pwnagotchi #289: [Bug]: pwnagotchi keeps freezing on "reading last session logs"

S-cpu3, 2025-01-04:
### What happened?

After Pwnagotchi boot, pwnagotchi freezes on "reading last session logs" and stays frozen on that screen. 
I am running the latest Jay image on a raspberry zero WH with a waveshare v4 e ink display. I have all default config.toml and default plugins. No changes. 

### Version

2.9.4

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

jayofelony, 2025-01-04:
No I don't know yet, reboot device.

S-cpu3, 2025-01-04:
I did still same result.

jayofelony, 2025-01-04:
Remove the logs, and reboot.

S-cpu3, 2025-01-04:
how do i remove the logs?

jayofelony, 2025-01-04:
Ssh in and remove logfiles.

S-cpu3, 2025-01-04:
How would removing the logs help with the freezing? I am just curious

jayofelony, 2025-01-04:
Well, it's stuck at reading logs so it wouldn't hurt to try at the very least. Maybe it's a very big file and it ran out of memory.

S-cpu3, 2025-01-04:
Just tried that to remove logs and reboot. It is now stuck on the first screen that pops up. Instead of reading last logs it says "Pretty fly 4 a Wi-FI" and "new day new pwns" and it freezes

jayofelony, 2025-01-04:
`sudo systemctl stop pwnagotchi && sudo pwnagotchi --debug`

I'm off to work, 6am here, I'll check back later.

S-cpu3, 2025-01-04:
Got it! Thank you!

S-cpu3, 2025-01-04:
Well the good news it gets past the "Pretty fly 4 a Wi-FI" and "new day new pwns" screen and starts searches for networks. The bad news is that it it back it freezing after 3-6 minutes of using it

jayofelony, 2025-01-04:
Now would be a good time to share the logs,might be something in it. Again reading logs?

S-cpu3, 2025-01-04:
It keeps saying that ui.fps = 0. How can i change it to 1?

jayofelony, 2025-01-04:
You are in manual mode probably, power it from the other micro usb port.

S-cpu3, 2025-01-04:
it is in the other USB port. It’s in AUTO mode

jayofelony, 2025-01-04:
No, that will only change refresh rate of the display. 0 is refresh on big changes, 1 is once per second.

S-cpu3, 2025-01-04:
Ohh got it! What other solutions could possibly fix this?

jayofelony, 2025-01-04:
Fix the reading logs issue?

dentaku65, 2025-01-04:
Can you please post the output of the command
`rfkill list`

S-cpu3, 2025-01-04:
0: phy0: Wireless LAN
        Soft blocked: yes
        Hard blocked: no
1: hci0: Bluetooth
        Soft blocked: no
        Hard blocked: no

dentaku65, 2025-01-04:
Wifi is software blocked do
`sudo rfkill unblock all`
Then
`sudo reboot`

S-cpu3, 2025-01-04:
would this help with the pwnagotchii freezing?

jayofelony, 2025-01-04:
> 0: phy0: Wireless LAN
>         Soft blocked: yes
>         Hard blocked: no
> 1: hci0: Bluetooth
>         Soft blocked: no
>         Hard blocked: no

That's odd, rfkill unblock all is run at boot.

S-cpu3, 2025-01-04:
will there be a patch in the future to fix this freezing?

wpa-2, 2025-01-04:
> will there be a patch in the future to fix this freezing?

Of course it will be looked at but in the mean time if you come across a fix then feel free to create a PR

jayofelony, 2025-01-04:
Did you ran apt upgrade by any chance?

S-cpu3, 2025-01-04:
I did that as well.

wpa-2, 2025-01-05:
> I did that as well.

Sadly that will brake stuff, your going to need to reflash. 
never run `sudo apt-get upgrade` update is always fine.

S-cpu3, 2025-01-05:
Oh yikes. Will reflash when a fix is out for the freezing. Thank you for letting me know!

wpa-2, 2025-01-05:
> Oh yikes. Will reflash when a fix is out for the freezing. Thank you for letting me know!

If setup correctly then yes of course. 
Always double check the steps here 
https://github.com/jayofelony/pwnagotchi/wiki

As this is now complete can you please close, hope fully you will be good going forward.

jayofelony, 2025-01-05:
> Oh yikes. Will reflash when a fix is out for the freezing. Thank you for letting me know! 

That is the fix, don't run upgrade.

S-cpu3, 2025-01-07:
> > Oh yikes. Will reflash when a fix is out for the freezing. Thank you for letting me know!
> 
> That is the fix, don't run upgrade.

This did not fix it. I just checked the reddit and people are having the same issues of the pwnagotchi freezing.

wpa-2, 2025-01-07:
> > > Oh yikes. Will reflash when a fix is out for the freezing. Thank you for letting me know!
> > 
> > 
> > That is the fix, don't run upgrade.
> 
> This did not fix it. I just checked the reddit and people are having the same issues of the pwnagotchi freezing.

Probably 2 reports of this and yours is one from what I can see, given the fact there's 1000s working fine then this really can't be an image issue. 

It's either hardware or user error, I could be wrong but from what I know thats normally the case.

S-cpu3, 2025-01-07:
so what should i check to make sure everything can work in order?

wpa-2, 2025-01-07:
> so what should i check to make sure everything can work in order?

When you say default everything, so its flash and boot and when you ssh in thats it> ?
what did debug say

S-cpu3, 2025-01-07:
I’ve posted the debugs logs before in this thread but it didn’t help much

S-cpu3, 2025-01-07:
No, I am on the latest version

wpa-2, 2025-01-07:
> No, I am on the latest version

Above says 2.9.4 is why I asked. 
Are you on the discord?

S-cpu3, 2025-01-07:
No I am not. I am on 2.9.4-2

wpa-2, 2025-01-07:
> No I am not. I am on 2.9.4-2

Talk me through your steps ?

S-cpu3, 2025-01-07:
I have a discord if you want me to send you messages through there. Just send me your username. 

> > No, I am on the latest version
> 
> Above says 2.9.4 is why I asked. Are you on the discord?

wpa-2, 2025-01-07:
> I have a discord if you want me to send you messages through there. Just send me your username.
> 
> > > No, I am on the latest version
> > 
> > 
> > Above says 2.9.4 is why I asked. Are you on the discord?

join the pwnagotchi group will be easier, tag me once your there

jalexandre0, 2025-01-12:
Hi there. Same happened to me when upgrded from 2.6 to 2.9.4-2. 
I reflash the sdcard, but bring config.toml and custom plugins from old install. 

In my case, bettercap passwords was diferent on config.toml these two files bellow. 

/usr/local/share/bettercap/caplets/pwnagotchi-auto.cap
/usr/local/share/bettercap/caplets/pwnagotchi-manual.cap

Changed the config.toml and my gotchi started to eat packages again. 

PS: This reddit thread from 4 years ago helped me to find the problem. 
https://www.reddit.com/r/pwnagotchi/comments/ig9m2i/solution_error_401_waiting_for_bettercap_api_to/

S-cpu3, 2025-01-12:
> Hi there. Same happened to me when upgrded from 2.6 to 2.9.4-2. I reflash the sdcard, but bring config.toml and custom plugins from old install. Im pretty sure my backup/restore messed things up. :)
> 
> In my case, bettercap passwords was diferent on config.toml these two files bellow.
> 
> /usr/local/share/bettercap/caplets/pwnagotchi-auto.cap /usr/local/share/bettercap/caplets/pwnagotchi-manual.cap
> 
> Changed the config.toml and my gotchi started to eat packages again.
> 
> PS: This reddit thread from 4 years ago helped me to find the problem. https://www.reddit.com/r/pwnagotchi/comments/ig9m2i/solution_error_401_waiting_for_bettercap_api_to/

Was there any version you had were it wasn't freezing. All i need to do is edit those files?

raffnex01, 2025-01-18:
I didn't read all the comments, but I have the same issue (2.9.4-2 and 2.9.5) with external WIFI antenna.

I can stop the freeze on "reading logs" when I unplug my external WIFI antenna and replug it. Then my pwnagotchi normally boots up and works like a charm.

Maybe this helps with your issue as well.

scansonic, 2025-01-19:
According to my observations, freezing on "last session logs reading" occurs when the device starts with not connected through Bluetooth or USB. When any connection is established, there is no freeze and everything works fine. Could anyone else check this behavior?
I noticed this behavior in the last three releases.

sakkarose, 2025-01-19:
This happens to randomly, still couldn't find the reason from the comments there.
I said randomly because sometime I just restart the device and it works again. I haven't connected it to my iphone through bluetooth.
But I'm gonna reflash to the newest and only transfer a pcap files as restore & reconfigure the new config files.

raffnex01, 2025-01-19:
> According to my observations, freezing on "last session logs reading" occurs when the device starts with not connected through Bluetooth or USB. When any connection is established, there is no [...]

Not in my case. I use a Raspberry Pi 4B without screen, so I have to be connected to my iPhone to see the error message.

Ulsterfry54, 2025-01-21:
Hi

This is a copy of my logfile when it does the same thing... not having a connection to google seems to initiate it.  I am on 2.9.5.1 on a Pi zero wh vanilla installation.

14:47:05 [ERROR] [Thread-5] : ConnectionError(MaxRetryError("HTTPSConnectionPool(host='www.google.com', port=443): Max retries exceeded with url: / (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0xac6be1d0>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution'))"))
(most recent call last):
 File "/usr/lib/python3/dist-packages/urllib3/connection.py", line 174, in _new_conn
   conn = connection.create_connection(
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
 File "/usr/lib/python3/dist-packages/urllib3/util/connection.py", line 73, in create_connection
   for res in socket.getaddrinfo(host, port, family, socket.SOCK_STREAM):
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
 File "/usr/lib/python3.11/socket.py", line 962, in getaddrinfo
   for res in _socket.getaddrinfo(host, port, family, type, proto, flags):
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[Errno -3] Temporary failure in name resolution

handling of the above exception, another exception occurred:

(most recent call last):
 File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 705, in urlopen
   httplib_response = self._make_request(
                      ^^^^^^^^^^^^^^^^^^^
 File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 388, in _make_request
   self._validate_conn(conn)
 File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 1050, in _validate_conn
   conn.connect()
 File "/usr/lib/python3/dist-packages/urllib3/connection.py", line 358, in connect
   self.sock = conn = self._new_conn()
                      ^^^^^^^^^^^^^^^^
 File "/usr/lib/python3/dist-packages/urllib3/connection.py", line 186, in _new_conn
   raise NewConnectionError(
<urllib3.connection.HTTPSConnection object at 0xac6be1d0>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution

handling of the above exception, another exception occurred:

(most recent call last):
 File "/usr/lib/python3/dist-packages/requests/adapters.py", line 489, in send
   resp = conn.urlopen(
          ^^^^^^^^^^^^^
 File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 789, in urlopen
   retries = retries.increment(
             ^^^^^^^^^^^^^^^^^^
 File "/usr/lib/python3/dist-packages/urllib3/util/retry.py", line 594, in increment
   raise MaxRetryError(_pool, url, error or ResponseError(cause))
HTTPSConnectionPool(host='www.google.com', port=443): Max retries exceeded with url: / (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0xac6be1d0>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution'))

handling of the above exception, another exception occurred:

(most recent call last):
 File "/home/pi/.pwn/lib/python3.11/site-packages/pwnagotchi/plugins/__init__.py", line 96, in process_events
   self.process_event(event_name, *args, **kwargs)
 File "/home/pi/.pwn/lib/python3.11/site-packages/pwnagotchi/plugins/__init__.py", line 85, in process_event
   callback(*args, **kwargs)
 File "/home/pi/.pwn/lib/python3.11/site-packages/pwnagotchi/plugins/default/ohcapi.py", line 73, in on_ui_update
   response = requests.get('https://www.google.com', timeout=5)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
 File "/usr/lib/python3/dist-packages/requests/api.py", line 73, in get
   return request("get", url, params=params, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
 File "/usr/lib/python3/dist-packages/requests/api.py", line 59, in request
   return session.request(method=method, url=url, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
 File "/usr/lib/python3/dist-packages/requests/sessions.py", line 587, in request
   resp = self.send(prep, **send_kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
 File "/usr/lib/python3/dist-packages/requests/sessions.py", line 701, in send
   r = adapter.send(request, **kwargs)
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
 File "/usr/lib/python3/dist-packages/requests/adapters.py", line 565, in send
   raise ConnectionError(e, request=request)
HTTPSConnectionPool(host='www.google.com', port=443): Max retries exceeded with url: / (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0xac6be1d0>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution'))
14:47:05 [INFO] [MainThread] : waiting for bettercap API to be available ...
14:47:06 [INFO] [MainThread] : waiting for bettercap API to be available ...
14:47:07 [INFO] [MainThread] : waiting for bettercap API to be available ...
14:47:08 [INFO] [MainThread] : waiting for bettercap API to be available ...
14:47:09 [INFO] [MainThread] : waiting for bettercap API to be available ...
14:47:10 [INFO] [MainThread] : waiting for bettercap API to be available ...
14:47:11 [INFO] [MainThread] : waiting for bettercap API to be available ...
14:47:12 [INFO] [Main

sakkarose, 2025-01-21:
> This happens to randomly, still couldn't find the reason from the comments there. I said randomly because sometime I just restart the device and it works again. I haven't connected it to my iphone [...]

It has been running fine so far atm. I had to do a config/plugin migration manually.

Marley-M, 2025-01-31:
I had this same issue, I reinstalled the [latest image](https://github.com/jayofelony/pwnagotchi/releases/tag/v2.9.5.2) and the problem stopped.

pkonkol, 2025-02-05:
Had this same problem on 2.9.5.3
After changing config and environment step by step it turned out that my brcmfmac is being broken after apt upgrade.
Something in here, turn off the auto-update for now.
```
pi@pwnagotchi:~ $ cat /var/log/apt/history.log

Start-Date: 2025-02-05  09:47:01
Commandline: apt upgrade -y
Requested-By: pi (1000)
Install: linux-headers-6.6.74+rpt-common-rpi:arm64 (1:6.6.74-1+rpt1, automatic), libcamera0.4:arm64 (0.4.0+rpt20250127-3, automatic), linux-headers-6.6.74+rpt-rpi-2712:arm64 (1:6.6.74-1+rpt1, automatic), linux-image-6.6.74+rpt-rpi-v8:arm64 (1:6.6.74-1+rpt1, automatic), linux-kbuild-6.6.74+rpt:arm64 (1:6.6.74-1+rpt1, automatic), linux-image-6.6.74+rpt-rpi-2712:arm64 (1:6.6.74-1+rpt1, automatic), linux-headers-6.6.74+rpt-rpi-v8:arm64 (1:6.6.74-1+rpt1, automatic)
Upgrade: rpicam-apps-lite:arm64 (1.5.3-1, 1.6.0-2), raspi-config:arm64 (20250129, 20250203), libpisp1:arm64 (1.0.7-1, 1.1.0-1), linux-headers-rpi-v8:arm64 (1:6.6.62-1+rpt1, 1:6.6.74-1+rpt1), linux-headers-rpi-2712:arm64 (1:6.6.62-1+rpt1, 1:6.6.74-1+rpt1), libcamera-ipa:arm64 (0.3.2+rpt20241119-1, 0.4.0+rpt20250127-3), linux-image-rpi-2712:arm64 (1:6.6.62-1+rpt1, 1:6.6.74-1+rpt1), libpisp-common:arm64 (1.0.7-1, 1.1.0-1), linux-image-rpi-v8:arm64 (1:6.6.62-1+rpt1, 1:6.6.74-1+rpt1), linux-libc-dev:arm64 (1:6.6.62-1+rpt1, 1:6.6.74-1+rpt1)
End-Date: 2025-02-05  09:53:35
```

wpa-2, 2025-02-05:
> Had this same problem on 2.9.5.3 After changing config and environment step by step it turned out that my brcmfmac is being broken after apt upgrade. Something in here, turn off the auto-update for [...]

You never run the upgrade command
Sadly you will need to backup and reflash.

Marley-M, 2025-02-06:
I have had this problem before and I have tried reflashing multiple times and it hasn't fixed the issue, I'm really confused as to what the problem could be.

S-cpu3, 2025-02-28:
Fixed: Change UI.FPS = 1

closed 2025-02-28 by S-cpu3