jayofelony/pwnagotchi #291: [Bug]: Key is stored in legacy trusted.gpg keyring

dentaku65, 2025-01-04:
### What happened?

Issuing apt update shows system alert 
```
sudo apt update
Hit:1 http://archive.raspberrypi.com/debian bookworm InRelease   
Get:2 http://raspbian.raspberrypi.com/raspbian bookworm InRelease [15.0 kB]
Fetched 15.0 kB in 9s (1681 B/s)                                                                                                                                                
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
W: http://raspbian.raspberrypi.com/raspbian/dists/bookworm/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.

To fix the issue 
`sudo cp /etc/apt/trusted.gpg /etc/apt/trusted.gpg.d/`

Ask for the next release the gpg key on a newer position 
```

### Version

2.9.4-2

### 3rd Party Hardware

No

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wpa-2, 2025-01-04:
`sudo apt update
Hit:1 http://deb.debian.org/debian bookworm InRelease
Hit:2 http://deb.debian.org/debian-security bookworm-security InRelease
Get:3 http://deb.debian.org/debian bookworm-updates InRelease [55.4 kB]
Hit:4 http://archive.raspberrypi.com/debian bookworm InRelease
Fetched 55.4 kB in 3s (18.8 kB/s)
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
`
Strange no issues here! 

Also don't run ` ``sudo apt upgarade```

closed 2025-01-05 by jayofelony

dentaku65, 2025-01-09:
Refresh image; confirm that the key is in the legacy position
[image]