jayofelony/pwnagotchi #295: [Bug]: UPS 0- using UPS Lite 1.3 and Plugin ups_lite

igor1019, 2025-01-10:
### Contact Details

igor1019@gmail.com

### What happened?

Hello, how are you?
I'm using the latest version of pwnagotchi and the activated ups_lite plugin only shows UPS 0- (it doesn't show the battery % even when using only the battery). I ran the i2detect command and the GPIOs are all correct in relation to the plugin code. What could it be?
Thanks in advance

[image]

### Version

2.9.4-2

### 3rd Party Hardware

No

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

TarasCZ, 2025-01-23:
What is output of your sudo i2cdetect -y 1, can u provide picture of how is it connected?

SgtStarSlayer, 2025-01-25:
any progress on text formatting for gps location overlapping with the text right above it or can we modify it ourselves?

Agnar319, 2025-02-01:
Use tweak-view to move it around however you want