jayofelony/pwnagotchi #296: [Bug]: Peers not being detected

gmccauley, 2025-01-11:
### Contact Details

gmccauley@gmail.com

### What happened?

Peers aren't being detected.  I've built 3 (2 on Pi4's and 1 on Pi02W) pwnagotchi's and none of them detect each other.

pwngrid-peer.log shows the api started and that advertisement is enabled:
```
2025-01-10 18:07:36 inf pwngrid v1.11.4 starting in peer mode ...
2025-01-10 18:07:38 inf /etc/pwnagotchi/id_rsa found
2025-01-10 18:07:38 inf started beacon discovery and message routing (0 known peers)
2025-01-10 18:07:38 inf peer CerealKiller@c8137048d2c63faf23cdbd05d1e992953ab05a9204902574b5f18c9c2e54c170 signaling is ready
2025-01-10 18:07:38 inf pwngrid api starting on 127.0.0.1:8666 ...
2025-01-10 18:08:41 inf peer advertisement enabled
```

I've ran wireshark from another device, and I don't see the beacons.  I've also run tcpdump directly on the pwnagotchi and don't see the beacons there either.  Looks like it's just not sending them out.

I tried to work through the code to try and debug but I don't know go.  Guess I'll be trying to teach myself enough of that this weekend.  LoL

### Version

2.9.4-2

### 3rd Party Hardware

No

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

jayofelony, 2025-01-11:
Try a higher number for auto-tune on extra channels. You can click the auto-tune plugin name when on the plugin page. It's at 3, try 9 or so.

gmccauley, 2025-01-11:
I was fixin to respond that didn't seem to help.  I even setup my kali laptop to sniff beacons and wasn't seeing any.  Decided to reboot all three instead of just using pwnkill and my kali laptop started seeing beacons and then my Pi3's started peering too.

The Pi02W never did send any beacons and my Pi3's only sent a handful and then 5-12 and then stopped.  Will revisit later.

gmccauley, 2025-01-12:
By sniffing beacons, I've come to the conclusion that beacons are sent for about a minute to a minute and a half after reboot and then stop.

avipars, 2025-01-19:
is "advertise" turned on (variable in personality)? ... it can be changed via auto_tune plugin too

gmccauley, 2025-01-24:
> is "advertise" turned on (variable in personality)? ... it can be changed via auto_tune plugin too

Yes, advertise is turned on.  Using another device, I'm able to see the beacons get sent for about a minute or two after reboot and then it stops.  It also seems like it stops receiving beacons around the same time.  My current working theory is that something is set to delay starting (maybe something with bettercap) and that is affecting the ability to send/receive beacons.  But I haven't really dug into it yet.

hb9eue, 2025-02-11:
Very similar issue to the one I reported a couple of months ago, but then only the Pi0W1 was affected. Never saw any beacons being sent and didn't manage to figure out the part of the code responsible to send them.