jayofelony/pwnagotchi #298: [Bug]: auto update not working 2.8.9 => 2.9.5

hb9eue, 2025-01-15:
### What happened?

Not sure what goes wrong.

My pwnagotchi is connected via USB tethering. The host provides NAT to the internet. I can resolve names an ping the internet from the pwnagotchi. The Auto-Update module is active.

### Version

2.9.4-2

### 3rd Party Hardware

No

### Relevant log output

```shell
root@pwn-hatchi:~# pwnagotchi --check-update
There is a new version available! Update from v2.8.9 to v2.9.5?
[Y/N] Y
ERROR:root:You should make sure auto-update is enabled!
Okay, give me a couple minutes. Just watch pwnlog while you wait.

Not sure if it wants to tell me that it tries to update. When I watch the usb network via tcpdump it does not look as it is transferring anything. i basically only see ntp packets being exchanged.

I also attempted to reboot and then try again, same result.

Any hints how to upgrade, without re-flashing?
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wpa-2, 2025-01-15:
Known issue, you need to backup and reflash due to a lot of changes.

hb9eue, 2025-01-15:
Got more logs out of it.

Well I guess i will backup /etc/pwnagotchi/ the ssh host keys and /boot/cmdline.txt and hopefully be fine after re-flashing.

```
13:22:46 [INFO] entering manual mode ...
13:22:47 [INFO] web ui available at http://[::]:8080/
13:23:30 [INFO] the last session lasted 5 hours, 43 minutes, 16 seconds (12 completed epochs, trained for 0), average reward:-0.049302600255990085 (min:-0.2 max:0.1)
13:23:36 [INFO] [update] checking for updates ...
13:23:36 [ERROR] error connecting to the pwngrid-peer service: list index out of range
13:23:36 [ERROR] [update] could not parse version from "bettercap -version": output=
/lib/arm-linux-gnueabihf/libc.so.6: version `GLIBC_2.32' not found (required by bettercap)
/lib/arm-linux-gnueabihf/libc.so.6: version `GLIBC_2.34' not found (required by bettercap)
13:23:43 [INFO] [update] checking for updates ...
13:23:43 [ERROR] error connecting to the pwngrid-peer service: list index out of range
13:23:43 [ERROR] [update] could not parse version from "bettercap -version": output=
/lib/arm-linux-gnueabihf/libc.so.6: version `GLIBC_2.32' not found (required by bettercap)
/lib/arm-linux-gnueabihf/libc.so.6: version `GLIBC_2.34' not found (required by bettercap)
13:23:50 [INFO] [update] checking for updates ...
```

wpa-2, 2025-01-15:
> Got more logs out of it.
>
> Well I guess i will backup /etc/pwnagotchi/ the ssh host keys and /boot/cmdline.txt and hopefully be fine after re-flashing.
>
> ```
> 13:22:46 [INFO] entering manual mode ... [...]

See my reply above

hb9eue, 2025-01-15:
Thank you, I have read the changes from past releases and indeed, it states you have to re-flash.

closed 2025-01-15 by hb9eue