jayofelony/pwnagotchi #299: [Bug]: i2c detection

bloop16, 2025-01-15:
### Contact Details

bloop16@hotmail.com

### What happened?

Hi,

Trying to get my pwnagotchi running with PiSugar S and Waveshare V4 display .
System is running fine, internet connection is working.
But since hours im not able to get i2c detecting the PisugarS board.

pwnagotchi 2.9.5
RaspberryPi Zero Wh (32bit)
PiSugarS

i2c is enabled in ``raspi-config``. 
Installed ``i2c-tools`` to check with ``i2cdetect -l``
i2c-1   i2c             bcm2835 (i2c@7e804000)                  I2C adapter
i2c-2   i2c             bcm2835 (i2c@7e805000)                  I2C adapter

Is it normal that it shows 2 different bus?

also checked ``i2cdetect 1`` and ``i2cdetect 2``
      0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

no device is showing up.

- already made a rpi firmware update
- disconnected from power for about 5min
- set ``dtparam=i2c_arm=on`` and  ``dtparam=i2c1=on`` in ``/boot/firmware/config.txt``

Is it possible that the PiSugarS Board isn`t compatible?'

Thanks for helping!

### Version

2.9.4-2

### 3rd Party Hardware

Yes

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

scansonic, 2025-01-15:
try run sudo i2cdetect -y 1

bloop16, 2025-01-15:
> try sudo apt-get install python3-smbus i2c-tools sudo i2cdetect -y 1

python3-smbus got installed to.
But didn`t make any change after reboot

hb9eue, 2025-01-15:
If you bought your PiSugar from Aliexpress advertised as 'works with Pwnagotchi' you might have got the same as I got which works, yes, but which does not communicate via i2c and only provides one button wired to gpio 3 so you can't even use it to shutdown on low battery. The only use of the button is to manually shutdown the raspi and also to turn it on.

bloop16, 2025-01-15:
> If you bought your PiSugar from Aliexpress advertised as 'works with Pwnagotchi' you might have got the same as I got which works, yes, but which does not communicate via i2c and only provides one [...]

Thanks! well bought on Amazon, but that could be the problem. Ordered an PiSugar 3 to check if it makes a difference.

rocketman887, 2025-01-19:
Hello, I am getting the same issue but on different setup, my setup is
Raspberry Pi Zero 2WH
Waveshare UPS_HAT_C
The battery is powering the Raspberry Pi but I just can't see the battery capacity
I bought it from reseller the hardware shouldn't be faulty right? Can I check in some other way if the HW is at fault?
Thank you!

rocketman887, 2025-01-19:
nevermind, I resolved my issue by flashing random version and after 5 repetition of flashing the newest image I got it working, it is probably the way I power it for the first time? I power through data and there is no led indication neither USB communication, then I need to unplug and plug again, there is the generating SSH keys part because I peeked monitor then it goes reset and I need to restart it once again to access the USB network

sschadwick, 2025-01-28:
Ran into a similar issue. Problem ended up being that Pi Zero 2WH (with factory headers) has excess flux on the bottom of the pins (while the original Zero W does not), resulting in a spotty connection. The pi was receiving power delivery but not i2c communication. Solution is to use a Zero 2W without headers and use hammer-in headers, or scrape the flux and re-solder.