jayofelony/pwnagotchi #300: [Bug]: Missing comma, nmcli invocation in bt-tether.py

hb9eue, 2025-01-15:
### What happened?

I noticed, that bt-tether was being set-up with a random ip address, not the one specified in config.toml

Line 44 is missing a comma on the line ipv4.dns

### Version

2.9.4-2

### 3rd Party Hardware

No

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

hb9eue, 2025-01-15:
btw, nmcli also complains about the dns servers. So not sure if that string is correctly formated. I just added one server - this works (well, not being able to connect anymore since updating from 2.8.9), but nmcli seems happy).

jayofelony, 2025-01-15:
Good eye!!

closed 2025-01-15 by jayofelony

akshat4165, 2025-01-18:
yeah now i got to know why BT-tether was not working, good work!

mihail4anov, 2025-01-19:
I have tried my luck with this new version, but after connection it immedeatly disconects.
Could not resolve smth - the error

wpa-2, 2025-01-19:
> I have tried my luck with this new version, but after connection it immedeatly disconects. Could not resolve smth - the error

Have you rebooted as Bluetooth only connects on startup.

mihail4anov, 2025-01-19:
> > I have tried my luck with this new version, but after connection it immedeatly disconects. Could not resolve smth - the error
> 
> Have you rebooted as Bluetooth only connects on startup.

I used commands as pair, connect, trust.

Aftect connect [mac] it changed BT to BT C, and the second after the error displays

wpa-2, 2025-01-19:
> > > I have tried my luck with this new version, but after connection it immedeatly disconects. Could not resolve smth - the error
> > 
> > 
> > Have you rebooted as Bluetooth only connects on startup.
> 
> I used commands as pair, connect, trust.
> 
> After connect [mac] it changed "BT -" to "BT C", the second after the error displays, and again "BT -"

you didnt say if you rebooted or not, check the logs it will help.

mihail4anov, 2025-01-19:
> you didnt say if you rebooted or not, check the logs it will help.

Thank you! I have decided to check the logs, and after power on it worked! So **reboot is necessary**

mihail4anov, 2025-01-19:
Ok, I have found the root of my problem. Old **config.toml** does not work))) Keep that in mind

akshat4165, 2025-01-22:
Did you solve it?
otherwise check nmcli and check if it is connected or disconnected