jayofelony/pwnagotchi #307: [Bug]: error with bt-tether.py

fmatray, 2025-01-19:
### What happened?

The bt-tether plugins is not able to up my bluetooth connection. 

the problem is in bt-tether.py while calling nmcli
`'ipv4.dns', '8.8.8.8;1.1.1.1;',`

the correct argument is **without semicolons** so it should be:
`'ipv4.dns', '8.8.8.8 1.1.1.1',`

I tried to change directly in the plugin as a quick (who said dirty? (^_^)  ) fix on my pwnagotchi and it works well.

### Version

2.9.4-2

### 3rd Party Hardware

No

### Relevant log output

```shell
A quick log to find the error.
21:12:17 [ERROR] [Thread-7] : [BT-Tether] Failed to connect to device: have you enabled bluetooth tethering on your phone? Command '['nmcli', 'connection', 'modify', 'MYPHONE Network', 'connection.type', 'bluetooth', 'bluetooth.type', 'panu', 'bluetooth.bdaddr', 'XX:XX:XX:XX:XX:XX', 'ipv4.method', 'manual', 'ipv4.dns', '8.8.8.8;1.1.1.1;', 'ipv4.addresses', '172.20.10.2', 'ipv4.gateway', '172.20.10.1', 'ipv4.route-metric', '100']' returned non-zero exit status 10.

I tried the command with sudo and I have this result:
Error: Failed to modify connection 'MYPHONE Network': ipv4.dns: 1. DNS server address is invalid
```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

wpa-2, 2025-01-19:
upgrade then, the problem is fixed on the new images.

fmatray, 2025-01-19:
Thanks, I just see the fix in the new update :-)

closed 2025-01-19 by fmatray