jayofelony/pwnagotchi #308: [Bug]: "Exception: error 400: 233 is not a valid wifi channel"

rsun0525, 2025-01-20:
### Contact Details

richard.d.sun@gmail.com

### What happened?

Figured out why my 6GHz channels were showing "disabled", basically I had to set the Wi-Fi Country (I set it to US), rebooted, and then I could see a lot more channels show up in the long, however, I get this error now:

2025-01-19 18:55:17,996] [INFO] [Thread-6] : Active: [6, 11, 40, 44, 52, 112, 149], Next scan: [6, 11, 40, 44, 52, 112, 149, 124, 233, 29, 4, 9, 60, 49, 185, 89, 153, 1, 201, 104, 9, 125], yet unscanned: 62 [3, 5, 7, 8, 10, 48, 56, 64, 100, 108, 116, 120, 128, 132, 140, 161, 165, 169, 173, 177, 1, 5, 13, 17, 25, 33, 37, 45, 53, 57, 65, 73, 77, 81, 85, 93, 97, 101, 105, 109, 117, 121, 129, 133, 145, 153, 157, 161, 165, 169, 173, 177, 181, 189, 193, 197, 205, 209, 213, 217, 225, 229]
[2025-01-19 18:55:18,460] [INFO] [Thread-4] : WPA_SEC: Internet connectivity detected. Uploading new handshakes to wpa-sec.stanev.org
[2025-01-19 18:55:18,467] [INFO] [MainThread] : error 400: 233 is not a valid wifi channel
[2025-01-19 18:55:18,470] [ERROR] [MainThread] : Error while setting wifi.recon.channels (error 400: 233 is not a valid wifi channel)
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/requests/models.py", line 971, in json
    return complexjson.loads(self.text, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/json/__init__.py", line 346, in loads
    return _default_decoder.decode(s)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/json/decoder.py", line 340, in decode
    raise JSONDecodeError("Extra data", s, end)
json.decoder.JSONDecodeError: Extra data: line 1 column 5 (char 4)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/.pwn/lib/python3.11/site-packages/pwnagotchi/bettercap.py", line 24, in decode
    return r.json()
           ^^^^^^^^
  File "/usr/lib/python3/dist-packages/requests/models.py", line 975, in json
    raise RequestsJSONDecodeError(e.msg, e.doc, e.pos)
requests.exceptions.JSONDecodeError: Extra data: line 1 column 5 (char 4)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/.pwn/lib/python3.11/site-packages/pwnagotchi/agent.py", line 160, in recon
    self.run('wifi.recon.channel %s' % ','.join(map(str, channels)))
  File "/home/pi/.pwn/lib/python3.11/site-packages/pwnagotchi/bettercap.py", line 118, in run
    return decode(r, verbose_errors=verbose_errors)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/.pwn/lib/python3.11/site-packages/pwnagotchi/bettercap.py", line 32, in decode
    raise Exception(err)
Exception: error 400: 233 is not a valid wifi channel

[pwnagotchi.log](https://github.com/user-attachments/files/18470940/pwnagotchi.log)

### Version

2.9.4-2

### 3rd Party Hardware

No

### Relevant log output

```shell
This is running v2.9.5.1, but there's no drop down under Version for me to choose this.
```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

rsun0525, 2025-01-20:
Had it occur on another one:

Traceback (most recent call last):
  File "/home/pi/.pwn/lib/python3.11/site-packages/pwnagotchi/agent.py", line 160, in recon
    self.run('wifi.recon.channel %s' % ','.join(map(str, channels)))
  File "/home/pi/.pwn/lib/python3.11/site-packages/pwnagotchi/bettercap.py", line 118, in run
    return decode(r, verbose_errors=verbose_errors)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/.pwn/lib/python3.11/site-packages/pwnagotchi/bettercap.py", line 32, in decode
    raise Exception(err)
Exception: error 400: 217 is not a valid wifi channel

[pwnagotchi.log](https://github.com/user-attachments/files/18470999/pwnagotchi.log)

rsun0525, 2025-01-20:
sudo iw phy0 info > output.txt

[output.txt](https://github.com/user-attachments/files/18471020/output.txt)

jayofelony, 2025-01-20:
Use raspi-config to change localization to US.

rsun0525, 2025-01-20:
Just to confirm, this is what you wanted me to do?

pi@pwnagotchi400:~ $ sudo raspi-config
Generating locales (this might take a while)...
  en_GB.UTF-8... done
  en_US.UTF-8... done
Generation complete.

rsun0525, 2025-01-20:
After making the above change, one of them got the error again:

[pwnagotchi.log](https://github.com/user-attachments/files/18478648/pwnagotchi.log)

[2025-01-20 09:11:54,403] [INFO] [MainThread] : [epoch 1] duration=00:00:48 slept_for=00:00:47 blind=0 sad=0 bored=0 inactive=0 active=1 peers=2 tot_bond=33.08 avg_bond=16.54 hops=1 missed=0 deauths=0 assocs=1 handshakes=0 cpu=2% mem=10% temperature=47C reward=0.10042918454935623
[2025-01-20 09:11:54,404] [INFO] [Thread-6] : Active: [6, 11, 36, 40, 44, 48, 56, 100, 112, 149], Next scan: [6, 11, 36, 40, 44, 48, 56, 100, 112, 149, 173, 65, 177, 2, 37, 85, 165, 117, 57, 53, 213, 177, 185, 81, 145], yet unscanned: 59 [1, 3, 5, 7, 9, 52, 60, 64, 104, 108, 116, 120, 128, 132, 136, 140, 144, 153, 157, 169, 1, 5, 9, 13, 17, 21, 25, 29, 33, 41, 49, 69, 73, 77, 89, 93, 97, 101, 105, 121, 125, 129, 133, 137, 141, 149, 153, 157, 165, 169, 173, 181, 189, 205, 209, 217, 225, 229, 233]
[2025-01-20 09:11:54,840] [INFO] [MainThread] : error 400: 213 is not a valid wifi channel
[2025-01-20 09:11:54,840] [ERROR] [MainThread] : Error while setting wifi.recon.channels (error 400: 213 is not a valid wifi channel)
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/requests/models.py", line 971, in json
    return complexjson.loads(self.text, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/json/__init__.py", line 346, in loads
    return _default_decoder.decode(s)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/json/decoder.py", line 340, in decode
    raise JSONDecodeError("Extra data", s, end)
json.decoder.JSONDecodeError: Extra data: line 1 column 5 (char 4)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/.pwn/lib/python3.11/site-packages/pwnagotchi/bettercap.py", line 24, in decode
    return r.json()
           ^^^^^^^^
  File "/usr/lib/python3/dist-packages/requests/models.py", line 975, in json
    raise RequestsJSONDecodeError(e.msg, e.doc, e.pos)
requests.exceptions.JSONDecodeError: Extra data: line 1 column 5 (char 4)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/.pwn/lib/python3.11/site-packages/pwnagotchi/agent.py", line 160, in recon
    self.run('wifi.recon.channel %s' % ','.join(map(str, channels)))
  File "/home/pi/.pwn/lib/python3.11/site-packages/pwnagotchi/bettercap.py", line 118, in run
    return decode(r, verbose_errors=verbose_errors)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/.pwn/lib/python3.11/site-packages/pwnagotchi/bettercap.py", line 32, in decode
    raise Exception(err)
Exception: error 400: 213 is not a valid wifi channel
[2025-01-20 09:11:55,782] [INFO] [Grid] : new peer pwnagotchi400@abb4b8efb488757899484aa804d1b129af771c13419596cccb8bf27db90015c0 detected (94334 encounters)

[pwnagotchi.log](https://github.com/user-attachments/files/18478649/pwnagotchi.log)

rsun0525, 2025-01-20:
Here's from another one:

[2025-01-20 09:11:33,785] [INFO] [MainThread] : [epoch 1] duration=00:00:47 slept_for=00:00:47 blind=0 sad=0 bored=0 inactive=2 active=0 peers=1 tot_bond=8.41 avg_bond=8.41 hops=0 missed=0 deauths=0 assocs=0 handshakes=0 cpu=2% mem=20% temperature=54C reward=-0.2
[2025-01-20 09:11:33,787] [INFO] [Thread-6] : Active: [6, 11, 40, 44, 52, 112, 149], Next scan: [6, 11, 40, 44, 52, 112, 149, 61, 10, 145, 93, 153, 29, 104, 56, 181, 165, 157, 225, 173, 185, 69], yet unscanned: 61 [2, 3, 4, 5, 7, 8, 9, 36, 48, 60, 100, 108, 124, 132, 140, 161, 169, 177, 1, 5, 9, 13, 17, 25, 33, 37, 41, 45, 49, 53, 57, 65, 73, 77, 81, 85, 89, 97, 101, 105, 109, 113, 117, 121, 125, 149, 157, 161, 169, 173, 177, 189, 193, 197, 201, 209, 213, 217, 221, 229, 233]
[2025-01-20 09:11:34,674] [INFO] [MainThread] : error 400: 181 is not a valid wifi channel
[2025-01-20 09:11:34,675] [ERROR] [MainThread] : Error while setting wifi.recon.channels (error 400: 181 is not a valid wifi channel)
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/requests/models.py", line 971, in json
    return complexjson.loads(self.text, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/json/__init__.py", line 346, in loads
    return _default_decoder.decode(s)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/json/decoder.py", line 340, in decode
    raise JSONDecodeError("Extra data", s, end)
json.decoder.JSONDecodeError: Extra data: line 1 column 5 (char 4)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/.pwn/lib/python3.11/site-packages/pwnagotchi/bettercap.py", line 24, in decode
    return r.json()
           ^^^^^^^^
  File "/usr/lib/python3/dist-packages/requests/models.py", line 975, in json
    raise RequestsJSONDecodeError(e.msg, e.doc, e.pos)
requests.exceptions.JSONDecodeError: Extra data: line 1 column 5 (char 4)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/.pwn/lib/python3.11/site-packages/pwnagotchi/agent.py", line 160, in recon
    self.run('wifi.recon.channel %s' % ','.join(map(str, channels)))
  File "/home/pi/.pwn/lib/python3.11/site-packages/pwnagotchi/bettercap.py", line 118, in run
    return decode(r, verbose_errors=verbose_errors)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/.pwn/lib/python3.11/site-packages/pwnagotchi/bettercap.py", line 32, in decode
    raise Exception(err)
Exception: error 400: 181 is not a valid wifi channel

[pwnagotchi.log](https://github.com/user-attachments/files/18478675/pwnagotchi.log)

jayofelony, 2025-01-20:
Localization, for wifi.

rsun0525, 2025-01-20:
That’s already been done, that’s how i got the 6ghz showing up in the first place. Which then led to this new error.

closed 2025-02-09 by rsun0525