jayofelony/pwnagotchi #312: A.I not working

grimeuk, 2025-01-20:
Hi, everything is working fine but it always seems to be in auto and has not yet gone into A.I. I’ve had it running for 2 hours but still no joy, does anybody know a solution please? Thank you!

grimeuk, 2025-01-20:
I must have been reading about older versions 🤦🏻‍♂️ thank you!

closed 2025-01-20 by grimeuk

wpa-2, 2025-07-05:
> This is just a fork of Pwnagotchi for whatever reason they kept the upstream name and have made changes to the project that no longer make it Pwnagotchi. Jay should rename instead of continuing to use the Pwnagotchi name.

why is it not pwnagotchi? 
AI become unstable so had to be removed, never heard so much crap in all my life.

wpa-2, 2025-07-06:
> It’s not Jay’s project to rename—it’s a fork, and that’s how open source works. The code may be open source, but the name is intellectual property. When you fork a project, especially if you’re introducing patches that aren’t accepted upstream, you’re expected to rename it. In this case, it’s not Pwnagotchi anymore—it’s just a fork.
> 
> Pwnagotchi, by definition, is an AI-driven project. When Jay removed the AI component, it fundamentally stopped being Pwnagotchi.
> 
> This has also caused confusion, as many people mistakenly believe Jay created Pwnagotchi, when in fact it was developed by evilsocket. As a result, credit is being misattributed to Jay by those who discover the project without knowing its origins.
> 
> It would be like someone taking the Ubuntu code and making changes and still calling it Ubuntu that's now how open source works and fundamentally violates the spirit of respecting the intellectual property of the project maintainer or person who created the code.

you can't blame jay because people no longer have the ability to read any more! 
its clearly stated who created pwnagotchi on the first page!