jayofelony/pwnagotchi #313: wps attack

Xanndiane, 2025-01-21:
could you add WPS Attacks or is this a difficult task? This could improve the efficiency

MimoAW, 2025-01-22:
That is an amazing idea! I'm not sure if it's possible though lol

MimoAW, 2025-01-22:
I have and believe me while bettercap over BT still worked I went on a walk around me village and 60% of them said WPS next to them

Xanndiane, 2025-01-22:
> > could you add WPS Attacks or is this a difficult task? This could improve the efficiency
> 
> Pretty sure wps is dead these days, then again it doesn't really help as your not there long enough to even get passed the first pin.
> 
> Have you ever used WPS attacks?

WPS vulnerable networks, at least in the post-Soviet countries are VERY numerous.

Xanndiane, 2025-01-22:
The only question is how it can be added. Pwnagotchi is just an add-on to bettercap, and bettercap can't do WPS attacks.

MimoAW, 2025-01-22:
Yeah I don't think bettercap can that's the issue if it could jayofelony would have already added it he might be able to do it somehow though he is very smart

TarasCZ, 2025-01-23:
It might be possible to create plugin using reaver, right?

C0d3-5t3w, 2025-01-27:
Does wifite still work?

MimoAW, 2025-01-27:
> Does wifite still work?
> 
> It's been 9 years since I put it to actual use and it wasn't so hot then.

I'm not sure but it doesn't look like we're getting support for it, which is fair because apparently it takes forever

avipars, 2025-05-05:
Best bet is to add an issue to the bettercap repository