jayofelony/pwnagotchi #315: ohcapi error?

Agnar319, 2025-01-23:
This release is running great, i just keep getting this error:

[2025-01-23 05:52:22,201] | [ERROR] | [Thread-10] : FileNotFoundError(2, 'No such file or directory')
  |   | Traceback (most recent call last):
  |   | File "/home/pi/.pwn/lib/python3.11/site-packages/pwnagotchi/plugins/__init__.py", line 96, in process_events
  |   | self.process_event(event_name, *args, **kwargs)
  |   | File "/home/pi/.pwn/lib/python3.11/site-packages/pwnagotchi/plugins/__init__.py", line 85, in process_event
  |   | callback(*args, **kwargs)
  |   | File "/home/pi/.pwn/lib/python3.11/site-packages/pwnagotchi/plugins/default/ohcapi.py", line 92, in on_ui_update
  |   | self._run_tasks(agent)
  |   | File "/home/pi/.pwn/lib/python3.11/site-packages/pwnagotchi/plugins/default/ohcapi.py", line 179, in _run_tasks
  |   | self.report.update(data={'reported': reported, 'processed_stations': processed_stations})
  |   | File "/home/pi/.pwn/lib/python3.11/site-packages/pwnagotchi/utils.py", line 702, in update
  |   | with ensure_write(self._path, 'w') as fp:
  |   | File "/usr/lib/python3.11/contextlib.py", line 137, in __enter__
  |   | return next(self.gen)
  |   | File "/home/pi/.pwn/lib/python3.11/site-packages/pwnagotchi/fs/__init__.py", line 19, in ensure_write
  |   | fd, tmp = tempfile.mkstemp(dir=path)
  |   | File "/usr/lib/python3.11/tempfile.py", line 496, in mkstemp
  |   | return _mkstemp_inner(dir, prefix, suffix, flags, output_type)
  |   | File "/usr/lib/python3.11/tempfile.py", line 395, in _mkstemp_inner
  |   | fd = _os.open(file, flags, 0o600)
  |   | FileNotFoundError: [Errno 2] No such file or directory: '/root/handshakes/tmpyznt7i_u'

Any Idea whats causing it?

MimoAW, 2025-01-23:
What version are you on?

wpa-2, 2025-01-23:
> This release is running great, i just keep getting this error:
>
> [2025-01-23 05:52:22,201] | [ERROR] | [Thread-10] : FileNotFoundError(2, 'No such file or directory')   |   | Traceback (most recent [...]

you need to upgrade pretty sure your using outdated pwnagotchi version.

Agnar319, 2025-01-23:
I'm on v2.9.5.2

wpa-2, 2025-01-24:
> I'm actually getting the same error on two different units, a 32bit and a 64bit, both on v2.9.5.2

/root/handshakes/ is the old diretory there now in /home/pi/handshakes/

closed 2025-01-28 by Agnar319