jayofelony/pwnagotchi #317: [Bug]: Cannot connect to phone via bluetooth

Dart-Alex, 2025-01-24:
### What happened?

Went trough the steps to pair & trust my phone (galaxy S24).
Somehow, the pi never connected to the phone.
Tried to manually connect (sudo bluetoothctl, connect <MAC>) and got this : 
[bluetooth]# connect <MAC>
Attempting to connect to <MAC>
[CHG] Device<MAC> Connected: yes
[CHG] Device <MAC> ServicesResolved: yes
Failed to connect: org.bluez.Error.NotAvailable br-connection-profile-unavailable
[CHG] Device <MAC> ServicesResolved: no
[CHG] Device <MAC> Connected: no

Couldn't find anything helpful on that error anywhere (other than for audio bt devices, but installing anything audio related onto the pi just made it behave like an audio device, which is not what we want here).

### Version

2.9.4-2

### 3rd Party Hardware

No

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

gmccauley, 2025-01-24:
Make sure bluetooth tethering is enabled.  This is the error I get when it's not.  Also my device (Pixel 7) disables bluetooth tethering every time it reboots, FYI.

wpa-2, 2025-01-24:
> Went trough the steps to pair & trust my phone (galaxy S24). Somehow, the pi never connected to the phone. Tried to manually connect (sudo bluetoothctl, connect ) and got this : [bluetooth]# connect Attempting to connect to [CHG] Device Connected: yes [CHG] Device ServicesResolved: yes Failed to connect: org.bluez.Error.NotAvailable br-connection-profile-unavailable [CHG] Device ServicesResolved: no [CHG] Device Connected: no
> 
> Couldn't find anything helpful on that error anywhere (other than for audio bt devices, but installing anything audio related onto the pi just made it behave like an audio device, which is not what we want here).
> 
> ### Version
> 2.9.5.2

secondly if you actually read the wiki not one place does it mewntion the connect command.

Dart-Alex, 2025-01-24:
Okay, so I was indeed barking up the wrong tree (don't have much experience with bluetooth shizzle).
Turns out when I filled in the mac address the first time, I mistyped it.
Fixed it, still not working.
Tried to run the nmcli command that the script runs manually, it told me two networks had the same name and I had to use id instead.
Don't know exactly what happened behind the scenes, but looking at nmcli, I had two networks with the same name (which caused the "on boot connect command" to fail since it uses the name as an identifier instead of the id).
I deleted the one that had the wrong mac address, now it works just fine.
Closing ;)

closed 2025-01-24 by jayofelony