jayofelony/pwnagotchi #323: [Bug]: brian.nn and brian.json

Marley-M, 2025-01-31:
### Contact Details

marley@tech-zone.cc

### What happened?

My Pwnagotchi doesn't have the AI files like brian.nn and brain.json.

The documentation on pwnagotchi.ai says they should be in /root but if I run `sudo ls /root` I only see bettercap.history, client_secrets.json, go, peers, and settings.yaml.

Have the AI files moved to a different location and the website hasn't been updated or do they not appear until the AI has finished training in which case does it need an internet connection to get the files?

I am running version 2.9.5.2 with Pi Zero W and a Waveshare V4 with a PiSugar S.

### Version

2.9.5.2

### 3rd Party Hardware

No

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

wpa-2, 2025-01-31:
we dont use .ai site as its VERY outdated now. 
AI was removed on this branch a while back for  various reasons. 
So the files your looking for are not there or used any more.

Marley-M, 2025-01-31:
Oh, where can I get an image that has the AI in it?

nasiralamreeki, 2025-01-31:
> Oh, where can I get an image that has the AI in it?
> 
> What website should I use instead? .org?

The original creator of GitHub has it as evilsocket. I think the AI version was more stable than what we have now and perhaps a bit faster scanning. Someday, more battery life, but the No AI builds introduced instability, probably going to be more releases for this to become a stable channel.

jayofelony, 2025-01-31:
Not a bug, closing.

closed 2025-01-31 by jayofelony