jayofelony/pwnagotchi #327: [Bug]: sudo pwnagotchi --wizard and -- check-update not working

Ryanwires, 2025-02-02:
### Contact Details

rmwebb915@gmail.com

### What happened?

sudo pwnagotchi (what ever command) results in "sudo: pwnagotchi: command not found" this happens for the wizard, version, donate, and update

### Version

2.9.5.2

### 3rd Party Hardware

Yes

### Relevant log output

```shell
can not get log due to issue bc "pwnlog" is also non functional
```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

wpa-2, 2025-02-02:
What login name have you used?

Ryanwires, 2025-02-03:
"pwn" which was set in raspberry pi imager

wpa-2, 2025-02-03:
> "pwn" which was set in raspberry pi imager

And that's the problem, You need to relash and read the wiki correctly.

Good luck

Ryanwires, 2025-02-03:
ahhh ok. thank you for the help!