jayofelony/pwnagotchi #329: Clear e-ink display on shutdown

Yorper, 2025-02-03:
I found this over on the Bjorn GitHub and thought it a very good idea.

https://github.com/infinition/Bjorn/issues/60

For those that don’t like following links, I’ve added a snippet below.

“
According to precaution https://github.com/infinition/Bjorn/issues/3 (https://www.waveshare.com/wiki/2.13inch_Touch_e-Paper_HAT_Manual):
'When using the e-Paper display, it is recommended that the refresh interval is at least 180s, and refresh at least once every 24 hours. If the e-Paper is not used for a long time, you should use the program to clear the screen before storing it. (Refer to the datasheet for specific storage environment requirements.)'

Apparently it will damage the screen (probably burn-in) if the screen is left with an image on it.

If we could get the screen to clear on shutdown from the web interface would be great. I'm used to Arduino stuff, I really don't know python at all, but according to the samples, you can clear the screen this way:

#Clear display: This function is used to clear the e-paper to white.
def Clear(self)

Also the precautions also mention:

For e-Paper displays that support partial refresh, please note that you cannot refresh them with the partial refresh mode all the time. After refreshing partially several times, you need to fully refresh EPD once. Otherwise, the display effect will be abnormal, which cannot be repaired!
Note that the screen cannot be powered on for a long time. When the screen is not refreshed, please set the screen to sleep mode or power off it. Otherwise, the screen will remain in a high voltage state for a long time, which will damage the e-Paper and cannot be repaired!
I don't know if these precautions are an issue, but I figured I would point them out just incase so our E-Paper screens don't get damaged.
“

SpaceLeftBlank, 2025-02-07:
I added that to my shutdown button on my PiSugar3 with the following .sh 

#!/bin/bash
sudo systemctl stop pwnagotchi
sudo pwnagotchi --clear
sudo shutdown -h now

Howardzi-nn, 2025-04-23:
> I added that to my shutdown button on my PiSugar3 with the following .sh
> 
> #!/bin/bash sudo systemctl stop pwnagotchi sudo pwnagotchi --clear sudo shutdown -h now

im using same thing, with **button-shim plugin**
[https://shop.pimoroni.com/products/button-shim]

as
`main.plugins.buttonshim.buttons.E.command = "systemd-run --unit=shutdown_clean /home/pi/shutdown_clean.sh"`

so cool 😄