jayofelony/pwnagotchi #332: Mac OS Connection and internet sharing guide

Ryanwires, 2025-02-09:
**Assigning interface IP and Internet sharing**
If the RNDIS/Ethernet Gadget does not automatically connect
System settings>Network, Click on the 3 dots in the lower right corner, Add service, and select the RNDIS/Ethernet Gadget in the drop down menu. Name it whatever you want really. now continue to the next section.

**Configuring the RNDIS/Ethernet**
Click on the System settings>Network>service>Details>TCP/IP. Here you will configure your IP and subnet mask. Under the drop down menu for Configure IPv4, select Manually. For the IP address now you will set it to 10.0.0.1. For the Subnet mask it will be 255.255.255.0 Next we will set the DNS servers, Click on the service>Details>DNS, Add one, enter 8.8.8.8 Add another, enter 1.1.1.1

You can now ssh into your Pwnagotchi

**To share internet to your Pwnagotchi**
Currently Mac OS’s built in internet sharing seems to break ssh. The way I have found around this is using EvilSocket’s connection sharing script ([Download here](https://github.com/evilsocket/pwnagotchi/blob/master/scripts/macos_connection_share.sh)) To use this script you need your interface number, to obtain such number, use ifconfig. Note down your main interface (Usually en0 sometimes en1 if you use ethernet and wifi) You will also need to find the interface number of your pwnagotchi, find the interface with the IP of 10.0.0.1. Note that number, for me en7.

Now, in a new terminal window. Type sudo sh. Drag the script you downloaded previously into the window, space once. Now enter the interface you're sharing from, for me en0 followed by a space the the interface you are sharing to. For me the command would look like 
sudo sh /Users/ryan/Downloads/macos_connection_share.sh en0 en7
Once you run this command you will be prompted for your password, enter said password and enter, you will see errors, 

sharing connecting from upstream interface en0 to usb interface en7 ...
net.inet.ip.forwarding: 0 -> 1
No ALTQ support in kernel
ALTQ related functions disabled
pf enabled
pfctl: Use of -f option, could result in flushing of rules
present in the main ruleset added by the system at startup.
See /etc/pf.conf for further details.

This is normal, it means it worked! Now you can go over to your Pwnagotchi and ping google.com to make sure everything has worked!

wpa-2, 2025-02-09:
There is also this 
https://pwnagotchi.org/getting-started/first-run-mac/index.html

jayofelony, 2025-02-09:
> **Assigning interface IP and Internet sharing**
> If the RNDIS/Ethernet Gadget does not automatically connect
> System settings>Network, Click on the 3 dots in the lower right corner, Add service, and [...]

Yes thank you. I'll add this asap to the wiki.

closed 2025-02-10 by jayofelony