jayofelony/pwnagotchi #336: Multi-Tethering

SpotlightForBugs, 2025-02-15:
I have multiple phones and I would like my pwnagotchi to connect to the default one if both are in range, but allow fallback ones too. This needs to support Mixed ios and Android

wpa-2, 2025-02-15:
So demanding fair play, as I mentioned learn to code and help it

SpotlightForBugs, 2025-02-15:
@wpa-2 
Same thing here. I would've assigned myself, but I have insufficient rights to do so.

[image]

Please update the https://github.com/jayofelony/pwnagotchi/blob/noai/.github/PULL_REQUEST_TEMPLATE.md if this is no longer needed.

Sniffleupagus, 2025-02-15:
I usually do the PR first, then raise the issue :)

fmatray, 2025-02-22:
It's technically possible it bt-tether but:
- BT drivers are buggy, so more tether means more changes to trigger a bug and have to reload kernel modules, services and connection.
- To implement correctly, each nm connection, nm device, bluetooth device status must be tracked properly + watchdog the kernel. It can lead to complex situations. Without that, chances increases to trigger that bug.
- With several phones, you need a switch and priority mechanism. NetworkManager have something like that already implemented but it must be used with autoconnect. 

With a better bluetooth stack it would be easy to rely on NetworkManager and Bluetooth services. With the actual stack... It's not an easy task.

avipars, 2025-03-11:
With the older version of BT Tethering this was possible and I was able to tether to 2 different android phones (not at the same time).

fmatray, 2025-03-11:
I'm trying to do it:
- One at the time, is quite easy.
- Several at the same time, it really depends on the configuration Raspi and phone. It's not possible to have 2 connections with the same gateway (phone's IP). So it needs manual configuration on raspberry and on all phones.