jayofelony/pwnagotchi #345: [Bug]: Pwndroid unavailable for this device

newrecycle, 2025-02-22:
### Contact Details

readerfreak@hotmail.ca

### What happened?

When I heard an android app for Pwnagotchi was released I got so excited I pulled out my old a11q, reflashed and re-rooted it, now when im about to buy the app it tells me its unavailable for my device :(

Could we get more info on the app, its requirements and maybe a timeline for a broader rollout?

### Version

2.9.5.3

### 3rd Party Hardware

Yes

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

newrecycle, 2025-02-22:
I should also add that the device is running Android 12, One UI 4.1 and a 4.9.227 kernel

jayofelony, 2025-02-22:
It should work from android 10 and up. But I don't know if itll work on a rooted device. I'll look up your device and see what restrictions it has.

newrecycle, 2025-02-22:
I just joined the discord. I could try unrooting and seeing if that solves it, but it doesnt seem to restrict me from downloading any other apps

newrecycle, 2025-02-22:
SOLUTION for those who have the same problem on a rooted device. 
Your device must have some level of play protect certification to buy and use Pwndroid (more about it here: [https://support.google.com/android/answer/7165974?hl=en](https://support.google.com/android/answer/7165974?hl=en) 

You must be using a rooting app that supports selecting certain apps and services to NOT be altered by the software, such as my case with Magisk. 

With Magisk, simply open its settings and enable "Enforce DenyList" and select the Google Play Services app (you must hit the show OS apps button) and select all of the services underneath it then reboot. This should allow you to purchase and download Pwndroid

closed 2025-02-22 by newrecycle