jayofelony/pwnagotchi #347: Question regarding Gdrivesync plugin

Agnar319, 2025-02-23:
Ive been setting up the gdrive plugin and, for the most part, its pretty straightforward, the only thing im unsure about is in its config.toml lines.

main.plugins.gdrivesync.enabled = false
main.plugins.gdrivesync.backupfiles = ['_**What goes here and in what format**_?']
main.plugins.gdrivesync.backup_folder = "PwnagotchiBackups"
main.plugins.gdrivesync.interval = 1

Any guidance is appreciated.
Thank You

jayofelony, 2025-02-23:
The absolute path you want to backup. Ie /etc/pwnagotchi for the entire folder or /etc/pwnagotchi/config.toml for just the config file in dictionary style

```
main.plugins.gdrivedync.backup_folders = [
"X",
"Y",
]
````

Agnar319, 2025-02-24:
Ok gotcha, thank you!

Agnar319, 2025-02-27:
While im asking silly questions about Gdrivesync. So i've got my unit backing up into Gdrive now perfectly, ( Auto-Backup backs up into /home/pi, so im backing up /home/pi into Gdrive, if that makes sense)  Now, my question is, If I was to have to re-flash or if i made an additional pwny, What would be the correct way to set up a new, or restore, using my .zip backup file?

closed 2025-03-09 by Agnar319