jayofelony/pwnagotchi #354: [Bug]: Pisugar unable to communicate

agrovista, 2025-02-28:
### Contact Details

ryan@brassmonkey.com

### What happened?

With the pisugar plugin enable and the data port in sue or the pisugar is off the pi zero is unable to communicate with the pisugar and the error is written to the logs every 5 seconds
this will cause pwnagotchi process to lock up the webui will not load pwnkill is unable to restart process etc 
the fix is to disable the pisugar plugin then hard reboot
the logs will also be huge and put the device reading logs from last session for a long time unless the logs are cleared.
This should fail gracefully 

I tested on a vanilla install of bookworm 
pisugar works fine and can communicate with pi zero 
once the data cable is plugged in address 0x75 is no longer able to see the pisugar

### Version

2.9.5.3

### 3rd Party Hardware

Yes

### Relevant log output

```shell
28:03 [ERROR] [MainThread] : No PiSugar device was found. Please check if the PiSugar device is powered on.
14:28:08 [ERROR] [MainThread] : No PiSugar device was found. Please check if the PiSugar device is powered on.
```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

jayofelony, 2025-02-28:
https://github.com/jayofelony/pwnagotchi/pull/342

Should work with a new version. When I have time, or just copy paste this new plugin version.

closed 2025-02-28 by jayofelony

agrovista, 2025-02-28:
> Thanks for the speedy reply !
> where do I need to paste the updated plugin ?
> /.pwn/lib/python3.11/site-packages/pisugar ?

never mind got it 
thank you all is good now

agrovista, 2026-02-27:
You will need to ssh in.

But this was an Issue I had with an older build of the software I would hope the current version does not have this issue any longer