jayofelony/pwnagotchi #355: [Bug]: Stuck in auto

Conroydowd2, 2025-03-01:
### What happened?

I'm on 2.9.5.3, and I have a fresh image, and it won't go into AI mode. If anyone knows a fix or needs more info, let me know; I've left it running for hours. It always boots into auto, never manual. 

I've seen people mention 
sudo apt install pip3 but it says sudo pip3 install --upgrade numpy
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.

    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.

    For more information visit http://rptl.io/venv

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
root@connigotchi:/home/pi#

### Version

2.9.5.3

### 3rd Party Hardware

No

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

wpa-2, 2025-03-01:
That's because AI was removed.

Conroydowd2, 2025-03-01:
> That's because AI was removed.

why?? I thought that was the whole point of what makes it special. Did it really do anything? Is it worth going to an older image.

wpa-2, 2025-03-01:
> > That's because AI was removed.
> 
> why?? I thought that was the whole point of what makes it special. Did it really do anything? Is it worth going to an older image.

nope and defo not! 
pwnagotchi has never run better.

closed 2025-03-01 by jayofelony