jayofelony/pwnagotchi #359: pwnagotchi not found

Diablox00, 2025-03-06:
I'm managed by systemd. Here are some basic commands.

    If you want to know what I'm doing, you can check my logs with the command
    - pwnlog
    - sudo pwnagotchi --wizard, to help set up a config.toml
    - sudo pwnagotchi --version, to check the current version
    - sudo pwnagotchi --donate, to see how you can donate to this project
    - sudo pwnagotchi --check-update, to see if there is a new version available

    If you want to know if I'm running, you can use
    sudo systemctl status pwnagotchi

    You can restart me using
    pwnkill

    You can learn more about me at https://pwnagotchi.org/
Last login: Thu Mar  6 07:27:26 2025 from 192.168.100.10
crus4d0r@raspberrypi:~ $ nano /etc/pwnagotchi/config.toml
crus4d0r@raspberrypi:~ $ sudo nano /etc/pwnagotchi/config.toml
crus4d0r@raspberrypi:~ $ cd /etc/pwnagotchi/conf.d/
crus4d0r@raspberrypi:/etc/pwnagotchi/conf.d $ ls
crus4d0r@raspberrypi:/etc/pwnagotchi/conf.d $ ls
crus4d0r@raspberrypi:/etc/pwnagotchi/conf.d $ ls -la
total 8
drwxr-xr-x 2 root root 4096 Jan 12 20:06 .
drwxr-xr-x 4 root root 4096 Mar  6 07:28 ..
crus4d0r@raspberrypi:/etc/pwnagotchi/conf.d $ cd ..
crus4d0r@raspberrypi:/etc/pwnagotchi $ sudo pwnagotchi --check-update
sudo: pwnagotchi: command not found
crus4d0r@raspberrypi:/etc/pwnagotchi $

wpa-2, 2025-03-06:
Read the guide it clearly says use pi user, just reflash.

Diablox00, 2025-03-07:
I figured it out, but anyways, still thank you.

closed 2025-03-07 by jayofelony

Diablox00, 2025-03-10:
I recently bought pwndriod, set up everything perfectly using BT
Teethering, connected  the  pwnagotchi to pwndriod, turned on share GPS but
it doesn't show any data or coordinates.
[image: image.png]