jayofelony/pwnagotchi #372: Can't connect to wifi for the life of me.

Canuckled, 2025-03-11:
Total noob, so please be patient. 

 Device: Pi Zero W
Screen: Waveshare 2.3 inch e-paper hat
jayofelon pwnagotchi-2.9.5.3-32bit 

When I first installed the release from the pwnagotchi.ai site, I was able to get everything working. Everything ran great for a few days, but then stopped. I then tried this release, had had it working for a bit,, but again,it just stopped. Since then, I have re-flashed I don't know how many times, with no luck. I went and bought a new sd card, thinking that might be the issue, but nothing. I searched the web more then I'd like to admit, and everything I've tried has not fixed my issue. I have no problem ssh'ing into my pi, but I just can't get WiFi nor Bluetooth to work. 

Any help would be greatly appreciated.

nasiralamreeki, 2025-03-11:
This is not the official Pwnagotchi repo this is a form that Jay created you probably want to follow the fork instructions. Since the fork doesn't use AI maybe good time for it to be renamed.

jayofelony, 2025-03-11:
Wifi cannot be used, as its used for monitoring. BT just follow the wiki step 4. Easy peazy.

Canuckled, 2025-03-11:
> Wifi cannot be used, as its used for monitoring. BT just follow the wiki step 4. Easy peazy.

Ugh, I've got to stop doing this stuff on very little sleep. 

Sorry for bothering.'

Thanks