jayofelony/pwnagotchi #373: [Bug]: pisugarx plugin

Usizi, 2025-03-13:
### What happened?

every time I try to enable the pisguarx login through the webui the web crashes and I cant re-access it until I re-setup my pwnagotchi

### Version

2.9.5.3

### 3rd Party Hardware

No

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

agrovista, 2025-03-24:
Have you installed the newer pisugar plugin ?
https://github.com/jayofelony/pwnagotchi/blob/noai/pwnagotchi/plugins/default/pisugarx.py

popeyesfx, 2025-03-25:
I have the same issue with the latest plugin

wpa-2, 2025-03-25:
> I have the same issue with the latest plugin

I use ti all the time and works fine, try ```debug```

wpa-2, 2025-03-25:
> I have the same issue with the latest plugin

the link above doesnt come with the image you flashed either you will need to replace it with the one on your pwnagotchi.

popeyesfx, 2025-03-28:
I replaced the plugin in the default directory, with the one from the link and have a icon on screen now, but the % never updates.
when on battery.

agrovista, 2025-05-06:
> I replaced the plugin in the default directory, with the one from the link and have a icon on screen now, but the % never updates. when on battery. log shows: [Thread-9] : [PiSugarX] PiSugar is not ready [2025-03-30 17:07:29,938] [INFO] [Thread-8 (_connect_device)] : No PiSugar device was found. Please check if the PiSugar device is powered on.

are you using a pi zero 2 ?
https://github.com/PiSugar/PiSugar/wiki/FAQ#1-why-pisugar-is-not-able-to-power-raspberry-pi-why-pisugar-23-cannot-be-detected-on-i2c-bus