jayofelony/pwnagotchi #377: [Bug]: Cannot get it to flash to microSD on two different computers and 3 different programs

Paranatural-Janitor, 2025-03-19:
### Contact Details

vixy91019@proton.me

### What happened?

balena etcher fails on end of verification, so does rufus, RPImager lets it run but wont boot, i'm at a loss, both were windows 11, all files come up as random strings of symbols, cannot find resources to move forward

### Version

2.9.5.3

### 3rd Party Hardware

No

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Paranatural-Janitor, 2025-03-19:
theres a wiki? im doing this off of a video

Paranatural-Janitor, 2025-03-19:
> Read the wiki for correct flashing options.

also im not seeing anything i haven't done, if this is a good faith atempt, maybe offer me a specific page, because im not finding the difference from anything ive tried

wpa-2, 2025-03-19:
All the videos are pointless 
https://github.com/jayofelony/pwnagotchi/wiki

the wiki is on the github linked above.

jayofelony, 2025-03-21:
User error. Closing.

closed 2025-03-21 by jayofelony