jayofelony/pwnagotchi #385: [Bug]: NOT INIT

daniel1rosso, 2025-03-27:
### Contact Details

danielalbertorosso@hotmail.com

### What happened?

I have a raspberry pi 5

I not use application, because is not running app with installed. I can see this error.
 sudo systemctl status pwnagotchi
● pwnagotchi.service - pwnagotchi Deep Reinforcement Learning instrumenting bettercap for WiFI pwning.
     Loaded: loaded (/etc/systemd/system/pwnagotchi.service; enabled; preset: enabled)
     Active: activating (auto-restart) since Wed 2025-03-26 23:53:32 GMT; 4s ago
       Docs: https://pwnagotchi.org
    Process: 5292 ExecStart=/usr/bin/pwnagotchi-launcher (code=exited, status=0/SUCCESS)
   Main PID: 5292 (code=exited, status=0/SUCCESS)
        CPU: 7ms

Mar 26 23:42:35 rpi systemd[1]: pwnagotchi.service: Scheduled restart job, restart counter is at 14.
Mar 26 23:42:35 rpi systemd[1]: Stopped pwnagotchi.service - pwnagotchi Deep Reinforcement Learning instrumenting bettercap for WiFI pwning..
Mar 26 23:42:35 rpi systemd[1]: Started pwnagotchi.service - pwnagotchi Deep Reinforcement Learning instrumenting bettercap for WiFI pwning..
Mar 26 23:42:35 rpi systemd[1]: pwnagotchi.service: Deactivated successfully.

pwnlog
tail: cannot open '/etc/pwnagotchi/log/pwnagotchi.log' for reading: No such file or directory
tail: no files remaining

sudo pwnagotc

### Version

2.9.5.2

### 3rd Party Hardware

No

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

wpa-2, 2025-03-27:
I'm guessing you either changed the username or your config is setup wrong /.
Reflash and follow this ?>? https://github.com/jayofelony/pwnagotchi/wiki

itschargeoclock, 2025-04-01:
im having issues with pi5, i flash sd (with both balena and raspery pi imager) dont change a thing no config jsut plug it in,  and it just goes red turns off instantly

i got it working again by playing with a few settings such as turning of the force host and attempting to force an LCD however i gave up on the screen and just tried to get it working

once it did sort of boot, windowns cannot recognise the device (code 43) i installend rndis and even tried updating through test mode removing the need for the signature. It works momentarily but the device reverts back to USB Composite