jayofelony/pwnagotchi #386: [Bug]: BT-Tether Crash

popeyesfx, 2025-03-30:
### Contact Details

popeyesfx@redih.co.uk

### What happened?

Have a crash in the BT-tether.py
Debug logs as follows:

[ERROR] [Thread-6] : [BT-Tether] Error with nmcli
[2025-03-30 16:39:04,520] [ERROR] [Thread-6] : [BT-Tether] Exception : Command '['nmcli', 'connection', 'modify', 'iphone Network', 'connection.type', 'bluetooth', 'bluetooth.type', 'panu', 'bluetooth.bdaddr', 'A8:4A:28:BF:B6:8A', 'connection.autoconnect', 'yes', 'connection.autoconnect-retries', '0', 'ipv4.method', 'manual', 'ipv4.dns', '8.8.8.8 1.1.1.1', 'ipv4.addresses', '175.20.10.2/24', 'ipv4.gateway', '172.20.10.1', 'ipv4.route-metric', '200']' returned non-zero exit status 10.

Have followed the guide on the site for IOS Phone.

Version of plugin is 1.4 copied from the repository.

### Version

2.9.5.2

### 3rd Party Hardware

No

### Relevant log output

```shell
[ERROR] [Thread-6] : [BT-Tether] Error with nmcli
[2025-03-30 16:39:04,520] [ERROR] [Thread-6] : [BT-Tether] Exception : Command '['nmcli', 'connection', 'modify', 'iphone Network', 'connection.type', 'bluetooth', 'bluetooth.type', 'panu', 'bluetooth.bdaddr', 'A8:4A:28:BF:B6:8A', 'connection.autoconnect', 'yes', 'connection.autoconnect-retries', '0', 'ipv4.method', 'manual', 'ipv4.dns', '8.8.8.8 1.1.1.1', 'ipv4.addresses', '175.20.10.2/24', 'ipv4.gateway', '172.20.10.1', 'ipv4.route-metric', '200']' returned non-zero exit status 10.
```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

popeyesfx, 2025-03-30:
Also noticed that I can connect via bluetooth on the phone but the pwnagotchi does not show connected. I did have version 1.2 of the plugin working for a short while before it failed too hence updating to 1.4

jayofelony, 2025-03-30:
I don't have a newer version released yet, so I cannot give support for something you've updated on your own behalf. Revert back to plugin version in 2.9.5.3 if you want my help.

closed 2025-03-31 by jayofelony