jayofelony/pwnagotchi #387: [Bug]: BT tethering does not connect to PAN

syntask, 2025-04-08:
### Contact Details

isaacneu@icloud.com

### What happened?

### Current behavior: 
Pwnagotchi successfully connects to iPhone's Bluetooth but does not connect to its associated PAN, rendering internet and web UI unusable. 

### Expected behavior: 
Pwnagotchi should automatically connect to PAN when bt-tether is enabled & configured properly.

### Steps to reproduce:
I paired my Pwnagotchi to my iPhone using bluetoothctl and config.toml as directed by the [wiki](https://github.com/jayofelony/pwnagotchi/wiki/Step-4-Customization/1012783dfdc562c88a0b0245e9d2a14d133349ef). My iPhone indicated that the Pwnagotchi was connected to Bluetooth but did not indicate that it was connected to Personal Hotspot. The Pwnagotchi displayed "BT C" onscreen. I was unable to connect to the Pwnagotchi's web UI from my phone. 

I ran the command `nmcli connection show` and saw a network listed as "Syntask's iPhone Network" but it was not connected to it. 

### Troubleshooting/temporary fix: 
1. Pair phone and Pwnagotchi as described in the [wiki](https://github.com/jayofelony/pwnagotchi/wiki/Step-4-Customization/1012783dfdc562c88a0b0245e9d2a14d133349ef). 
2. SSH into Pwnagotchi via USB.
3. Run `nmcli connection show` and look for a network with type "Bluetooth." For iPhone users the network will be called "<hostname> Network" e.g. "Syntask's iPhone Network"
4. Run `sudo nmcli connection up "<network>"` replacing `<network>` with the name of your phone's network from the previous step.
5. Run `ip a` to list the network interfaces currently in use, and verify that `bnep0` is now listed.
6. Run `sudo nmcli connection modify "<network>" connection.autoconnect yes` replacing `<network>` with the name of your phone's network as before. Now the Pwnagotchi will automatically connect to the phone's PAN network anytime it is available via bluetooth. 

### Hardware:
- Raspberry Pi Zero 2W running 2.9.5.2 64-bit image
- Waveshare v4
- PiSugar 3
- iPhone 13 Pro running iOS 18.3.2

### Version

2.9.5.2

### 3rd Party Hardware

No

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

mustang5269, 2025-04-11:
I'm running an rpi4 with latest image and I am having the same issue. Bluetooth connects on iPhone, pwnagotchi shows btc but never uses the hotspot.

syntask, 2025-04-11:
> I'm running an rpi4 with latest image and I am having the same issue. Bluetooth connects on iPhone, pwnagotchi shows btc but never uses the hotspot.

Good to know. Did you try the temporary fix steps I put above?

wpa-2, 2025-04-23:
> same issue with hw `rpi zero w` sw `2.9.5.3` phone `iPhone 16 Pro iOS 18.1`
> 
> edit: i have problem with share internet, so i figure out my route
> 
> ```
> pi@howard:~ $ ip route
> default via 10.0.0.1 dev usb0 proto static metric 100 
> default via 172.20.10.1 dev bnep0 proto dhcp src 172.20.10.12 metric 750 
> 10.0.0.0/24 dev usb0 proto kernel scope link src 10.0.0.2 metric 100 
> 172.20.10.0/28 dev bnep0 proto kernel scope link src 172.20.10.12 metric 750 
> ```
> 
> there was two default routes, usb0 and bnep0 so we need to change default gateway

They will be different ?>

Howardzi-nn, 2025-04-24:
> They will be different ?>

like, they are different
**10.0.0.1, metric 100 | 172.20.10.1, metric 750**
but for some reason, the rpi tries to connect via **usb0, metric 100** (and its "stuck" there)

what works (for me) is to delete the **usb0** or higher the metrics. 
maybe someone more knowledgeable will comment this issue

Gregory88PL, 2025-05-02:
same issue
hw rpi zero w
sw 2.9.5.3
phone Samsung Fold5 Android 15

Gregory88PL, 2025-05-03:
I noticed that if I first connect via Bluetooth and then via data cable, the internet works normally.

taboo8614, 2025-06-06:
After following your steps all of my connection issues were resolved. Thanks for posting about this!