jayofelony/pwnagotchi #391: Problem with Bettercap when restarting the Pwnagotchi

Moineau54, 2025-04-16:
Hello, 

I currently am experiencing a problem with Bettercap, as it is semingly not working.
```
waiting for bettercap API to be available ...
```

```
sudo systemctl status bettercap

● bettercap.service - bettercap api.rest service.
     Loaded: loaded (/etc/systemd/system/bettercap.service; enabled; preset: enabled)
     Active: activating (auto-restart) (Result: exit-code) since Wed 2025-04-16 15:16:22 BST; 3s ago
       Docs: https://bettercap.org
    Process: 1141 ExecStart=/usr/bin/bettercap-launcher (code=exited, status=1/FAILURE)
   Main PID: 1141 (code=exited, status=1/FAILURE)
        CPU: 489ms
```

How can I fix this ?

Moineau54, 2025-04-16:
I think I fixed it on my part.
What I previously did was enabling Wlan preconfiguration when flashing the Pi. Disabling it seems to have fixed it.

jayofelony, 2025-04-17:
In my wiki guide I told you to click "No, no custom settings".

closed 2025-04-17 by jayofelony