jayofelony/pwnagotchi #394: [Bug]: waiting for bettercap API to be available

Sharkrider120, 2025-04-24:
### Contact Details

sharkrider120

### What happened?

every time i try to use auto mode on my pwnagotchi it never works and in pwn log it just says  
05:33:53 [INFO] [MainThread] : waiting for bettercap API to be available ...
over and over again

### Version

2.9.5.2

### 3rd Party Hardware

No

### Relevant log output

```shell
05:27:31 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:27:32 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:27:34 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:27:35 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:27:36 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:27:37 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:27:39 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:27:40 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:27:41 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:27:42 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:27:43 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:27:44 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:27:45 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:27:46 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:27:47 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:27:48 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:27:49 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:27:51 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:27:52 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:27:53 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:27:54 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:27:55 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:27:56 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:27:57 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:27:58 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:27:59 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:28:00 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:28:01 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:28:02 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:28:03 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:28:05 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:28:06 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:28:07 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:28:08 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:28:09 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:28:10 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:28:11 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:28:12 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:28:13 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:28:14 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:28:15 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:28:17 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:28:18 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:28:19 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:28:20 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:28:21 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:28:22 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:28:23 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:28:24 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:28:25 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:28:26 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:28:27 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:28:28 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:28:29 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:28:31 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:28:32 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:28:33 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:28:34 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:28:35 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:28:36 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:28:37 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:28:38 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:28:39 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:28:40 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:28:41 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:28:42 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:28:44 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:28:45 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:28:46 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:28:47 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:28:48 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:28:49 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:28:51 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:28:52 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:28:53 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:28:54 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:28:55 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:28:56 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:28:58 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:28:59 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:29:00 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:29:01 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:29:02 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:29:03 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:29:04 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:29:05 [INFO] [MainThread] : waiting for bettercap API to be available ...
05:29:06 [INFO] [MainThread] : waiting for bettercap API to be available ...
```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Sharkrider120, 2025-04-24:
im gunna try a reflash

Sharkrider120, 2025-04-24:
well that fix it

Sharkrider120, 2025-04-24:
never mind its still not working right

pi@pwnagotchi:~ $ ifconfig
lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 44  bytes 6082 (5.9 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 44  bytes 6082 (5.9 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

usb0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 10.0.0.2  netmask 255.255.255.0  broadcast 10.0.0.255
        ether da:ae:02:18:b8:43  txqueuelen 1000  (Ethernet)
        RX packets 1074  bytes 212499 (207.5 KiB)
        RX errors 0  dropped 4  overruns 0  frame 0
        TX packets 1004  bytes 776949 (758.7 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether b8:27:eb:1d:81:38  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0mon: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        unspec 00-00-00-00-00-00-E8-3B-00-00-00-00-00-00-00-00  txqueuelen 1000  (UNSPEC)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

pi@pwnagotchi:~ $ ifconfig
lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 368  bytes 591483 (577.6 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 368  bytes 591483 (577.6 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

usb0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 10.0.0.2  netmask 255.255.255.0  broadcast 10.0.0.255
        ether da:ae:02:18:b8:43  txqueuelen 1000  (Ethernet)
        RX packets 1349  bytes 265696 (259.4 KiB)
        RX errors 0  dropped 4  overruns 0  frame 0
        TX packets 1096  bytes 791271 (772.7 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

jayofelony, 2025-04-24:
Using an external wifi? Is it properly connected?

Sharkrider120, 2025-04-24:
I'm using a raspberry pi 0

Sharkrider120, 2025-04-24:
I don't know if it went bad or something

Sharkrider120, 2025-05-08:
so i tried it with another pi zero and it also had it wlan device just randomly vanish from ifconfig

Sharkrider120, 2025-05-08:
im just going to flash raspberry pi os to it to see if the wlan0 module is working

wpa-2, 2025-05-09:
> Ok so it's something with the image I guess because raspberry pi os runs just fine with wifi

The image works fine, my guess is you skipped a step on the wiki. 
most of the time its due to changing the user form PI

Dx3One, 2025-07-03:
same here, i updated today my SD-Card with 2.9.5.3 gotchi stoped working 
pwnlog says: : waiting for bettercap API to be available

ifconfig show no wlan0 interface
and raspi-config also failed with message : no 'wifi' device

i think missing firmware in the last images

Dx3One, 2025-07-03:
> same here, i updated today my SD-Card with 2.9.5.3 gotchi stoped working pwnlog says: : waiting for bettercap API to be available
> 
> ifconfig show no wlan0 interface and raspi-config also failed with message : no 'wifi' device
> 
> i think missing firmware in the last images

ive compared the image info files and see that base image is the same as last working 2.9.5

TechsavvyCNC, 2025-07-03:
I flashed image 2.9.4-2 and it's working now.

Dx3One, 2025-07-03:
i can confirmed 2.9.4-2 is working 

[image]

theflyingwombat, 2025-10-24:
> same here, i updated today my SD-Card with 2.9.5.3 gotchi stoped working pwnlog says: : waiting for bettercap API to be available
> 
> ifconfig show no wlan0 interface and raspi-config also failed with message : no 'wifi' device
> 
> i think missing firmware in the last images

May or may not be relevant but I just did a reflash with 2.9.5.3 and had the same issues with no wlan0 interface and the bettercap API messages.  It started working once I set the WLAN country in the raspi-config and rebooted.