jayofelony/pwnagotchi #397: [Bug]: can't connect after changing IP

lands39, 2025-05-01:
### What happened?

pwnagotchi defaults to the 10.0.0.0/24 subnet, this makes it unusable for me as my LAN uses this subnet and it would cause a conflict. i changed the pwnagotchi's IP to 10.69.0.2 as this doesn't conflict with anything, but i cannot connect to it anymore. it worked briefly once, but after i rebooted it stopped and no longer appears to be functional.

### Version

2.9.5.2

### 3rd Party Hardware

No

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

tholtom, 2025-05-25:
You can get around the network clash.  I had the same problem, so I started an in memory instance of Ubuntu, with no external network connections.  You can then set up the 10.0.0.1/24 network for usb0 in ubuntu and connect that way.

lands39, 2025-05-25:
i do not have the ability to use IOMMU on my hypervisor, so i cannot pass through network or USB devices. i wa later able to change the IP through networkmanager and i got USB to cooperate

tholtom, 2025-05-26:
How did you do it?  I'd be interested to know, where exactly you made the changes to assign a different static IP address.