jayofelony/pwnagotchi #399: Manage multiple Bluetooth devices (two or more android phones, PC, etc...)

halo981, 2025-05-05:
I'd like to configure multiple bluetooth devices (two or more android phones, two or more iPhone, two PC, etc...).

With this feature i can use different devices at different times (one device at time).

Into the configuration.toml file i tried to insert more android devices, but pwnagotchi wouldn't start.

It is possible to configure multiple bluetooth devices, without the limitation of one android and one iphone?

Thanks

wpa-2, 2025-05-05:
> > > I'd like to configure multiple bluetooth devices (two or more android phones, two or more iPhone, two PC, etc...).
> > > With this feature i can use different devices at different times (one device at time).
> > > Into the configuration.toml file i tried to insert more android devices, but pwnagotchi wouldn't start.
> > > It is possible to configure multiple bluetooth devices, without the limitation of one android and one iphone?
> > > Thanks
> > 
> > 
> > I control 3 x pwnagotchi with one android phone if that's what's your looking to do then yes very easy.
> 
> No i'd like to associate one pwnagotchi to multiple bluetooth devices (two android phones, one PC).
> 
> When i need to connect to pwnagotchi i can use one of the devices that i have associated to pwnagotchi.
> 
> I tried to insert two android phones into configuration.toml, but pwnagotchi wouldn't start.

I dont think its possible sadly, whats the need for it though ?

halo981, 2025-05-05:
> I dont think its possible sadly, whats the need for it though ?

For example if i need to connect to pwnagotchi web page with another phone (if my first phone is out of battery), or if i need to connect with a notebook to make some configuration changes (with terminal) on the fly