jayofelony/pwnagotchi #403: Permission denied

Pokesh-f, 2025-05-14:
Hello im trying to make the pwnagotchi work so i use a video guide : https://www.youtube.com/watch?v=xY6cJHUl36I

But i cannot for a life of me get in to the ssh login 

[image]

I tried raspberry, changeme, pwnagotchi, pwnagotch

wpa-2, 2025-05-14:
Ignore all videos and start here 
https://github.com/jayofelony/pwnagotchi/wiki

read and read again!

wpa-2, 2025-05-14:
> > Ignore all videos and start here https://github.com/jayofelony/pwnagotchi/wiki
> > read and read again!
> 
> Omg this has its own wiki omggg, i will read it!!! Thank you so much man idk why someone does a tutorial and has link on totaly diff instalation <3

Always use the source :)

vervs3, 2025-05-18:
I have the same issue( though followed the guide. @Pokesh-f have you resolved it?

Pokesh-f, 2025-05-18:
> I have the same issue( though followed the guide. [@Pokesh-f](https://github.com/Pokesh-f) have you resolved it?

Yes just follow the guide my biggest mistake was that i used the raspberry first as pihole and i had settings in pi imager to have custom password.

P3NG00, 2025-11-20:
Make sure to follow the guide. Default password will be 'raspberry'. Make sure you aren't overriding the user/password with raspberry pi imager's custom configuration settings.