jayofelony/pwnagotchi #405: [Bug]: bt-tether.py fails to bt tether via wifi.

PapiJalopy, 2025-05-17:
### What happened?

Your rewritten `bt-tether.py` plugin fails to tether over phones WIFI connection.

Previous `bt-tether.py` plugin, written originally I believe by "dadav", had no issues tethering both data and WIFI seamlessly, while this rewritten one fails to resolve while phone is connected to WIFI. 

This requires me to disable WIFI and use only a data connection to establish network tether with the rpi.

For anyone needing this functionality back I recommend replacing `bt-tether.py` found in -

```/home/pi/.pwn/lib/python3.11/site-packages/pwnagotchi/plugins/default/```

with the original `bt-tether.py`, found in evilsocket's repo or other pwnagotchi forks and repositories.

You will need to re-enable the `bt-tether.py` plugin after replacing it, add the options below to `/home/pi/.pwn/lib/python3.11/site-packages/pwnagotchi/defaults.toml` and adjust them in webui after a reboot, and/or manually add and adjust them in `/etc/pwnagotchi/config.toml`

```
main.plugins.bt-tether.enabled = false

main.plugins.bt-tether.devices.android-phone.enabled = false
main.plugins.bt-tether.devices.android-phone.search_order = 1
main.plugins.bt-tether.devices.android-phone.mac = ""
main.plugins.bt-tether.devices.android-phone.ip = "192.168.44.44"
main.plugins.bt-tether.devices.android-phone.netmask = 24
main.plugins.bt-tether.devices.android-phone.interval = 1
main.plugins.bt-tether.devices.android-phone.scantime = 10
main.plugins.bt-tether.devices.android-phone.max_tries = 10
main.plugins.bt-tether.devices.android-phone.share_internet = false
main.plugins.bt-tether.devices.android-phone.priority = 1

main.plugins.bt-tether.devices.ios-phone.enabled = false
main.plugins.bt-tether.devices.ios-phone.search_order = 2
main.plugins.bt-tether.devices.ios-phone.mac = ""
main.plugins.bt-tether.devices.ios-phone.ip = "172.20.10.6"
main.plugins.bt-tether.devices.ios-phone.netmask = 24
main.plugins.bt-tether.devices.ios-phone.interval = 5
main.plugins.bt-tether.devices.ios-phone.scantime = 20
main.plugins.bt-tether.devices.ios-phone.max_tries = 0
main.plugins.bt-tether.devices.ios-phone.share_internet = false
main.plugins.bt-tether.devices.ios-phone.priority = 999
```

You will also need to either comment out, or delete -

```
#main.plugins.bt-tether.enabled = false
#main.plugins.bt-tether.phone-name = "" # name as shown on the phone i.e. "Pwnagotchi's Phone"
#main.plugins.bt-tether.mac = ""
#main.plugins.bt-tether.phone = "" # android or ios
#main.plugins.bt-tether.ip = "" # 192.168.44.2 android / 172.20.10.2 ios
```

until this regression is fixed.

 

### Version

2.9.5.2

### 3rd Party Hardware

No

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

wpa-2, 2025-05-17:
The current bt-tether works fine without issues if setup right, theres no popint using old outdated plugins on the newer image. 
https://github.com/jayofelony/pwnagotchi/wiki/Step-4-Customization#bt-tethering

PapiJalopy, 2025-05-17:
Hi @wpa-2,
I believe I was fairly clear in my original explanation, but let me clarify again in case there was any misunderstanding.

The issue isn’t about improper setup. The rewritten bt-tether.py plugin just simply does not support bluetooth tethering over a phone’s Wi-Fi connection. It only resolves correctly when the phone is using mobile data. This behavior is a regression from the original script (from dadav’s version), which handled bluetooth tethering over both data and Wi-Fi seamlessly.

Claiming that “it works fine if set up right” misses the point. This isn’t a configuration problem. It’s a limitation in the current implementation that wasn’t present in the previous one.

Until this is addressed, users who rely on Wi-Fi-based Bluetooth tethering will encounter this issue, and reverting to the older script is the only practical workaround atm.

jayofelony, 2025-05-17:
It works for me.. does your isp router allow internet sharing?

PapiJalopy, 2025-05-17:
Works with "old" bt_tether.py. Not working with rewritten bt_tether.py. Android.