jayofelony/pwnagotchi #409: [Bug]: Wifi Recon Flipped, stops finding networks and reboots

RaceBX2, 2025-05-24:
### What happened?

Hi, after disabling wifi in config.txt, adding an antenna and rebooting, I  get the error Wifi Recon Flipped and it reboots. This continues regardless of how many reboots occur. When it runs, it's finding networks, but it's very short lived. Panda Wireless PAOUD antenna. Any ideas?

### Version

2.9.5.2

### 3rd Party Hardware

No

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

wpa-2, 2025-05-24:
fix_services plugin is only for internal wifi so if your using external you can disable it.

RaceBX2, 2025-05-25:
Still happening with fix_services disabled.

wpa-2, 2025-05-25:
> Still happening with fix_services disabled.

maybe post some logs, how do you expect people to help with out them.

reqsyszeroize, 2025-07-29:
I had the same problem after adding my AWUS036ACS antenna but it wasn't rebooting, only "Wifi recon flipped" appearing every 10 secondes.
Disabling "fix_services" in my config made it disapear, tyy !