jayofelony/pwnagotchi #411: Scrolling ssid instead of static

Curious-Keeper, 2025-06-17:
**Is your feature request related to a problem? Please describe.**
As of now the agent.py has a static displayed ssid next to the lifetime count. the issue arises specifically on the waveshare 2.13 builds when you encounter networks or aps with longer than 15 characters in them. I know this isn't something that happens all the time, but it does happen. when it happens the ssid layers over all other plugin information across the bottom of the screen, resulting in a black blob of text.

**Describe the solution you'd like**
update site-packages/pwnagotchi/agent.py to include scrolling text variables

` # Scrolling text variables 
        self._scroll_text = None
        self._scroll_position = 0
        self._scroll_pause = 0
        self._scroll_direction = 1  # 1 for right, -1 for left
        self._scroll_speed = 0.5  # seconds between scrolls
        self._scroll_pause_time = 3  # seconds to pause at start/end
        self._max_display_length = 15  # maximum characters to display at once
`

---
and 

         `if self._last_pwnd is not None:
            # Handle scrolling text for long SSIDs
            if len(self._last_pwnd) > self._max_display_length:
                current_time = time.time()
                
                # Initialize scrolling if this is a new SSID
                if self._scroll_text != self._last_pwnd:
                    self._scroll_text = self._last_pwnd
                    self._scroll_position = 0
                    self._scroll_pause = current_time + self._scroll_pause_time
                    self._scroll_direction = 1
                
                # Handle scrolling logic
                if current_time >= self._scroll_pause:
                    if self._scroll_direction == 1:  # Moving right
                        if self._scroll_position + self._max_display_length >= len(self._scroll_text):
                            self._scroll_direction = -1
                            self._scroll_pause = current_time + self._scroll_pause_time
                        else:
                            self._scroll_position += 1
                            self._scroll_pause = current_time + self._scroll_speed
                    else:  # Moving left
                        if self._scroll_position <= 0:
                            self._scroll_direction = 1
                            self._scroll_pause = current_time + self._scroll_pause_time
                        else:
                            self._scroll_position -= 1
                            self._scroll_pause = current_time + self._scroll_speed
                
                # Get the current visible portion of the text
                visible_text = self._scroll_text[self._scroll_position:self._scroll_position + self._max_display_length]
                txt += ' [%s]' % visible_text
            else:
                txt += ' [%s]' % self._last_pwnd
`

**Describe alternatives you've considered**
Alternative to scrolling display would be a simple character limitation so that it doesn't cover the other text

jayofelony, 2025-06-17:
What plugin stuff do you have at the bottom than? I've set ssid length to be 11 or 12 characters longif I remember correctly. What version are you on?

Curious-Keeper, 2025-06-17:
[image]

on latest. using assoc and deauth plugins on bottom.

Curious-Keeper, 2025-06-18:
I don't have a screenshot of the overlapping, and I have already changed my agent.py to the proposed revision. before it would overlap the assoc and assoc count. Specifically if it located a wifi hotspot or access point, since I test mine mostly with other devices that send beacons and AP's to make sure it all works. Those generally are long threads, not the typical default wifi names.

Before I changed it, long network names would go across almost the entire bottom of the screen. This may also be because I have hard coded a lot of my plugin positions so tweak view doesn't really work for me.

closed 2025-06-20 by jayofelony