jayofelony/pwnagotchi #424: [Bug]: Wavshare2in13_V4 won't display

hjscott7690, 2025-08-07:
### Contact Details

hjscott7690@hotmail.com

### What happened?

no matter what i put in, i keeps replacing with dummydisplay.py during restart, just to be clear the file doens't change just whats in debug

[ui.display]
enabled = true
rotation = 0
type = 'waveshare2in13_V4'

### Version

2.9.5.2

### 3rd Party Hardware

No

### Relevant log output

```shell
Traceback (most recent call last):
  File "/usr/bin/pwnagotchi", line 8, in <module>
    sys.exit(pwnagotchi_cli())
             ^^^^^^^^^^^^^^^^
  File "/home/pi/.pwn/lib/python3.11/site-packages/pwnagotchi/cli.py", line 322, in pwnagotchi_cli
    display = Display(config=config, state={'name': '%s>' % pwnagotchi.name()})
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/.pwn/lib/python3.11/site-packages/pwnagotchi/ui/display.py", line 12, in __init__
    super(Display, self).__init__(config, hw.display_for(config), state)
  File "/home/pi/.pwn/lib/python3.11/site-packages/pwnagotchi/ui/view.py", line 51, in __init__
    self._layout = impl.layout()
                   ^^^^^^^^^^^^^
  File "/home/pi/.pwn/lib/python3.11/site-packages/pwnagotchi/ui/hw/dummydisplay.py", line 28, in layout
    lw, lh = fonts.Small.getbbox("W")
    ^^^^^^
ValueError: too many values to unpack (expected 2)
```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

wpa-2, 2025-08-08:
type debug in terminal 
Also it should be waveshare_4

hjscott7690, 2025-08-08:
Sorry i meant Wavshare2in13_V3 not Wavshare2in13_V4 but i changed it to waveshare_3 and works great. thank you for the assist

wpa-2, 2025-08-09:
> Sorry i meant Wavshare2in13_V3 not Wavshare2in13_V4 but i changed it to waveshare_3 and works great. thank you for the assist

Can you close then then now its sorted please.

closed 2025-10-02 by hjscott7690