jayofelony/pwnagotchi #426: [Bug]: Not able to SSH over USB after initial installation on MacOS

introkun, 2025-08-19:
### Contact Details

introkun@gmail.com

### What happened?

Steps:
- flash image to sdcard (no custom settings) Rpi Zero 2 W
- insert sdcard and boot
- usb connected to Mac
- manually configured ip to 10.0.0.1
- pinging 10.0.0.1 or trying ssh

Actual result:
- no ping also NDIS says "connected"
- not able to ssh

### Version

2.9.5.2

### 3rd Party Hardware

No

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

wpa-2, 2025-08-21:
https://pwnagotchi.org/getting-started/first-run-mac/index.html

did you try this.

wpa-2, 2025-08-21:
> > https://pwnagotchi.org/getting-started/first-run-mac/index.html
> > did you try this.
> 
> Yes. I was able to finally connect from Steam Deck

Can this be closed then :)

introkun, 2025-08-24:
Same cable works for steam deck...
What can I provide for this issue? How to get the logs?

billygr, 2025-08-26:
Same issue here with MacOS Sequoia 15.6.1.

L2 works (means arp exists) 
? (10.0.0.1) at [dedacted]on en9 ifscope permanent [ethernet]
? (10.0.0.2) at [ee:5d:dedacted] on en9 ifscope [ethernet]

ping fails, ssh fails

JackThursby252, 2025-09-02:
Just made a new one on 15.6.1, everything works (web+ssh). Including internet sharing via usb. 

Bluetooth tested on latest ios also works with web +ssh. 

Imho config issue on the client devices

Delfairen, 2026-01-11:
Same issue here with MacOS Sequoia 15.7.2

Steps:

flash 32 bit 2.9.5.3 image to SD card (no custom settings) Raspberry pi Zero W
insert SD Card and boot
USB connected to Mac on Data port
Wi-Fi turned off as it is a 10.0.0.0/24 network on the Wi-Fi
manually configured RNDIS IP to 10.0.0.1 255.255.255.0 DNS 8.8.8.8
Set service order so RDNIS is first

Actual result:

no ping
no ssh

Version
2.9.5.3

I will give any logs you want if you can tell me what you need.

Delfairen, 2026-01-11:
I do think this is a MacOS issue but I have not been able to confirm this.