jayofelony/pwnagotchi #428: [Bug]:

Antoine1-byte, 2025-08-28:
### Contact Details

antoine.del62@gmail.com

### What happened?

Describe the bug
Wlan0 is not available whatever pwnagotchi image is used

Steps to reproduce the behavior:

- Install pwnagotchi
- ifconfig

Expected behavior

- Expectation after installation is that when ssh the pwnagotchi, I shall have a wlan0 interface to get it working but only lo and usb0 are visible.

Screenshots

[image]

As you can see, the Pi Zero is properly identified via cat /proc/cpuinfo

Environment (please complete the following information):

- different versions have been tested
- pi zero W 1.1

Additional context

- Strange thing is that if I install raspiOS 2025-05-13-raspios-bookworm-armhf.img the Pi Zero W is identified as a Raspberry Pi 3 Model B Rev 1.2 and that with this OS, wlan interface is working

[image]

I'm trying to fix this issue since couple of days, tried some "force_board_rev=" and other stuff but nothing is working.
Fact that with RaspiOS the wlan is properly working, my conclusion is that it is not link to hardware but somewhere in the software...

Thanks in advance for any tips you would have

Antoine

### Version

2.9.5.2

### 3rd Party Hardware

No

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

jayofelony, 2025-08-30:
The software has no issues, there is something about the hardware. What wifi chipset is it?

`sudo dmesg | grep brcmfmac`

eplas, 2025-08-31:
Hello, same problem with two boards of Pi Zero W V1.1, here is sudo dmesg | grep brcmfmac:

[   31.930131] brcmfmac: loading out-of-tree module taints kernel.
[   32.012145] brcmfmac: F1 signature read @0x18000000=0x1541a9a6
[   32.271795] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[   32.272179] usbcore: registered new interface driver brcmfmac
[   33.220160] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[   33.221898] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[   33.222103] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
[   33.239632] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Aug  7 2017 00:46:29 version 7.45.41.46 (r666254 CY) FWID 01-f8a78378
[   38.545493] brcmfmac: brcmf_vif_add_validate: Attempt to add a MONITOR interface...
[   38.545549] brcmfmac: brcmf_mon_add_vif: brcmf_mon_add_vif called
[   38.545563] brcmfmac: brcmf_mon_add_vif: Adding vif "wlan0mon"
[   84.837293] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled

And lsmod | grep brcmfmac:
brcmfmac              307200  0
brcmutil               16384  1 brcmfmac
cfg80211              860160  1 brcmfmac

Any ideas?

Antoine1-byte, 2025-09-01:
> The software has no issues, there is something about the hardware. What wifi chipset is it?
> 
> `sudo dmesg | grep brcmfmac`

Hello,

Here are the results:
pi@pwnagotchi:~ $ sudo dmesg | grep brcmfmac
[   34.588679] brcmfmac: loading out-of-tree module taints kernel.
[   34.646103] usbcore: registered new interface driver brcmfmac
[   80.541874] usbcore: deregistering interface driver brcmfmac
[   83.643477] usbcore: registered new interface driver brcmfmac
[  126.761220] usbcore: deregistering interface driver brcmfmac
[  129.052009] usbcore: registered new interface driver brcmfmac

pi@pwnagotchi:~ $ lsmod | grep brcmfmac
brcmfmac              307200  0
brcmutil               16384  1 brcmfmac
cfg80211              860160  1 brcmfmac

chaosjester, 2025-12-24:
I have the same issue on Pi Zero W v1.1 and 1.3

pi@pwnerer:~ $ sudo dmesg | grep brcmfmac
[   30.717819] brcmfmac: loading out-of-tree module taints kernel.
[   30.762196] usbcore: registered new interface driver brcmfmac
[   75.042850] usbcore: deregistering interface driver brcmfmac
[   77.652874] usbcore: registered new interface driver brcmfmac
[  116.700384] usbcore: deregistering interface driver brcmfmac
[  118.453370] usbcore: registered new interface driver brcmfmac
[  156.789580] usbcore: deregistering interface driver brcmfmac
[  158.343090] usbcore: registered new interface driver brcmfmac
[  167.842589] usbcore: deregistering interface driver brcmfmac
[  169.379800] usbcore: registered new interface driver brcmfmac
[  196.704220] usbcore: deregistering interface driver brcmfmac
[  198.258583] usbcore: registered new interface driver brcmfmac
[  236.582929] usbcore: deregistering interface driver brcmfmac
[  238.211844] usbcore: registered new interface driver brcmfmac
[  276.565333] usbcore: deregistering interface driver brcmfmac
[  278.189932] usbcore: registered new interface driver brcmfmac
[  316.585023] usbcore: deregistering interface driver brcmfmac
[  318.141558] usbcore: registered new interface driver brcmfmac
[  356.549956] usbcore: deregistering interface driver brcmfmac
[  358.359244] usbcore: registered new interface driver brcmfmac
[  396.796607] usbcore: deregistering interface driver brcmfmac
[  398.346946] usbcore: registered new interface driver brcmfmac
[  436.783170] usbcore: deregistering interface driver brcmfmac
[  438.347998] usbcore: registered new interface driver brcmfmac
[  476.815473] usbcore: deregistering interface driver brcmfmac
[  478.368231] usbcore: registered new interface driver brcmfmac
[  516.811776] usbcore: deregistering interface driver brcmfmac
[  518.389828] usbcore: registered new interface driver brcmfmac
[  556.883503] usbcore: deregistering interface driver brcmfmac
[  558.675168] usbcore: registered new interface driver brcmfmac
[  596.965140] usbcore: deregistering interface driver brcmfmac
[  598.504233] usbcore: registered new interface driver brcmfmac

pi@pwnerer:~ $ lsmod | grep brcmfmac
brcmfmac              307200  0
brcmutil               16384  1 brcmfmac
cfg80211              860160  1 brcmfmac